chipKIT32/chipKIT-core #11: Incomplete 1.6.x api integration

ricklon, 2015-07-17:
Incomplete 1.6.x api integration. Currently, this build identifies as Arduino 1.5 compatible. But we need to finish the changes to get all the 0023 code update to 1.6+. 

This needs the help of the community and it's faster to work here than in MPIDE core.

--Rick

ricklon, 2015-09-14:
A lot of work has been done on this. We are almost there. More documentation and testing is needed.

EmbeddedMan, 2016-01-06:
This is a  longer term task, and will need to broken out into separate sub-issues, since the Arduino libraries/core are a moving target, and we need to  not only make sure our API matches Arduino's, but that our functionality also matches theirs.