chipKIT32/chipKIT-core #23: Windows download doesn't work - for e92169a build of core

EmbeddedMan, 2015-07-26:
I compile the blink sketch for Fubarino SD with Arduino 1.6.5, and then try to download it to my Fubarino SD in bootloader mode (this works fine with MPIDE) and get the following errors (see end):

Again, it seems suspect that only 484 bytes are used for the sketch. And my COM24 works just fine with MPIDE.

*Brian

C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-g++ -O2 -c -mno-smart-io -w -fno-exceptions -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\Blink.cpp -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\Blink.cpp.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-gcc -O2 -c -mno-smart-io -w -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD\Board_Data.c -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\Board_Data.c.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-g++ -O2 -g1 -c -Wa,--gdwarf-2 -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\cpp-startup.S -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\cpp-startup.S.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-g++ -O2 -g1 -c -Wa,--gdwarf-2 -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\crti.S -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\crti.S.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-g++ -O2 -g1 -c -Wa,--gdwarf-2 -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\crtn.S -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\crtn.S.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-g++ -O2 -g1 -c -Wa,--gdwarf-2 -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\pic32_software_reset.S -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\pic32_software_reset.S.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-g++ -O2 -g1 -c -Wa,--gdwarf-2 -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\vector_table.S -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\vector_table.S.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-gcc -O2 -c -mno-smart-io -w -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\exceptions.c -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\exceptions.c.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-gcc -O2 -c -mno-smart-io -w -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\HardwareSerial_cdcacm.c -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\HardwareSerial_cdcacm.c.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-gcc -O2 -c -mno-smart-io -w -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\HardwareSerial_usb.c -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\HardwareSerial_usb.c.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-gcc -O2 -c -mno-smart-io -w -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\pins_arduino.c -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\pins_arduino.c.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-gcc -O2 -c -mno-smart-io -w -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\task_manager.c -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\task_manager.c.o 
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-gcc -O2 -c -mno-smart-io -w -ffunction-sections -fdata-sections -g3 -mdebugger -Wcast-align -fno-short-double -fframe-base-loclist -mprocessor=32MX795F512H -DF_CPU=80000000L -DARDUINO=10605 -D_PIC32_FUBARINO_SD -DMPIDEVER=16777998 -DMPIDE=150 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32 -IC:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\variants\Fubarino_SD C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32\cores\pic32\WInterrupts.c -o C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp\WInterrupts.c.o 
[... cut ...]
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/compiler/pic32-tools/bin/pic32-bin2hex -a C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp/Blink.cpp.elf 

Sketch uses 484 bytes (0%) of program storage space. Maximum is 520,192 bytes.
Global variables use 32 bytes of dynamic memory.
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core/pic32/tools/bin/pic32prog.exe -d COM24 -b 115200 C:\Users\Brian\AppData\Local\Temp\build9069320794225117443.tmp/Blink.cpp.hex 
COM24: Cannot open
Programmer for Microchip PIC32 microcontrollers, Version 2.0.147

    Copyright: (C) 2011-2015 Serge Vakulenko
No target found.

ricklon, 2015-07-28:
Can you add a screen shot of the directory location?

EmbeddedMan, 2015-07-28:
Rick,

Which directory do you mean?

The path to the chipKIT core is :
C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32

The path to Arduino is :
C:\Program Files (x86)\Arduino

The path where the sketch is getting built is :
C:\Users\Brian\AppData\Local\Temp\build7740872639994275538.tmp

*Brian

ricklon, 2015-07-28:
So weird. Those are correct.

--Rick

On Tue, Jul 28, 2015 at 10:10 AM, Brian Schmalz <notifications@github.com>
wrote:

> C:\Users\Brian\Documents\Arduino\hardware\chipKIT-core\pic32
>

-- 
Co-founder
Fair Use Building and Research (FUBAR) Labs
http://fubarlabs.org

EmbeddedMan, 2015-07-28:
Yeah, that's really strange that it works for you. Do you know anyone else
who's tried it and gotten it to work on any flavor of Windows?

*Brian

EmbeddedMan, 2015-07-28:
Could it be a problem with my MPIDE or Arduino configuration files? Do those matter?

*Brian

ricklon, 2015-07-28:
I had it detect the Fubarino SD with the Teensy com port driver on a fresh machine that had Teensy on it. The programmer couldn't talk to the port. I think I've got a lead to follow.

ricklon, 2015-07-29:
My new guess is that it's a pic32prog error for Windows.

EmbeddedMan, 2015-07-29:
Were you able to get the latest version of pic32prog into the Windows core
zip file?

ricklon, 2015-07-29:
I've compiled the the latest version from source and it has the same
problem. I'm not confident that I compiled it improperly.

ricklon, 2015-07-30:
Ok. I just tried from a fresh Windows 10 install. I installed Arduino. Created hardware folder. Then installed the chipKIT-core. Then I enabled USB for Virtual Box. I had then installed the stk500v2 drivers. I started Arduino and it programmed blink and serial communication just fine.

Currently, I'm guessing there is a problem with incorrectly matching Teensy drivers and stk500v2 drivers.

EmbeddedMan, 2015-07-30:
I don't get how that explains my problem though. I can't program anything -
I'm not in a VM, I'm not using Teensy or it's drivers, etc. I have a
perfectly working MPIDE system, and when I try to use the current Windows
chipKIT core, nothing will program. It feels to me like there's something
else going on here. But I'm not sure how to start debugging it from this
point.

*Brian

ricklon, 2015-07-30:
I did find the problem on two Windows 7 machines at work. They both had Teensy on them. I've been working with them to test as well. What I think needs to happen is testing with a know good hex file and various versions of pic32prog. So it's not really testing the core but focusing on the uploader.

ricklon, 2015-07-30:
Teensy thing seems to not be an issue. 

I'll have to confirm I can compile pic32prog and test it with know good hex.

rianders, 2015-07-30:
Below is a link to a "TestWinIssue.bat" that has the local test also the sketch it runs is included.
https://s3.amazonaws.com/chipkit/chipkit-core/tests/WinIssueTest.zip

This is code that's working on a Windows 7 machine with stk500v2 drivers. I watch it load and reset in the device manager.

To test;
Put Fubarino in programming mode. Then run  testpic32prog.bat . 

I'm going to bring it to the machines that are failing the test. 

I'll let you know how it goes. Maybe, there is pic32prog verbose mode. I'm looking for that.

EmbeddedMan, 2015-07-30:
Rick, I tried with your zip file. I only have a MAX32, not a Fubarino SD
with me right now, but things are still failing (and not because of the
board difference):

Here's what I get:

D:\WinIssueTest>testpic32prog.bat

D:\WinIssueTest>pic32prog.exe -d COM11 -b 115200 WinIssueTest.cpp.hex
Programmer for Microchip PIC32 microcontrollers, Version 2.0.147
    Copyright: (C) 2011-2015 Serge Vakulenko
COM11: Cannot open

No target found.

D:\WinIssueTest>

Now, I have no problem opening COM11 with MPIDE, or any terminal emulator.
So this is a problem with pic32prog.exe.

*Brian

EmbeddedMan, 2015-07-30:
Rick, I FOUND THE FREEEEEEEEKING ISSUE.

It all has to do with com port numbers.

I deleted every com port on my machine, then plugged in my MAX32, and it
enumerated as COM 4 (rather than COM 11). I then updated the bat file to
reflect this, and now I get:

D:\WinIssueTest>testpic32prog.bat

D:\WinIssueTest>pic32prog.exe -d COM4 -b 115200 WinIssueTest.cpp.hex
Programmer for Microchip PIC32 microcontrollers, Version 2.0.147
    Copyright: (C) 2011-2015 Serge Vakulenko
      Adapter: STK500v2 Bootloader
 Program area: 1d000000-1d1fffff
    Processor: Bootloader
 Flash memory: 2048 kbytes
  Boot memory: 80 kbytes
         Data: 25160 bytes
        Erase: done
Program flash: ########################### done
 Verify flash: ########################## done
 Program rate: 2998 bytes per second

D:\WinIssueTest>

In other words, completely successful.

What this means is that pic32prog.exe does NOT recognize higher com ports.
This is a common problem with programs that are not written using 'windows
native' type methods, as some languages/libraries were only able to access
COM ports from 1-4, or 1-9, etc.

So this needs to get logged on pic32prog's issue tracker, and fixed. Until
then, this won't work for us, as many people have chipKIT boards on higher
COM ports.

*Brian

EmbeddedMan, 2015-07-30:
Rick,

As per https://github.com/sergev/pic32prog/issues/12, I have solved my
issue with pic32prog by using the full "windows" com port naming system, as
per
https://support.microsoft.com/en-us/kb/115831

What this means is that you'll need to modify the core somehow to add the
"\\.\" in front of all COM port names, when running under Windows only.
Then pic32prog will work just fine, and this issue will be solved. In other
words, instead of

pic32prog.exe -d COM256 -b 115200 WinIssueTest.cpp.hex

we need it to be

pic32prog.exe -d \\.\COM256 -b 115200 WinIssueTest.cpp.hex

*Brian

ricklon, 2015-07-30:
I can modify and test with that. It's a setting I think can be adjusted.

ricklon, 2015-07-30:
1. I have more symptoms from testing and interesting results. I want to document them here:
Windows 7 in device manager reported coms 3-11 in use. If I assigned the board a port number lower than 10 and then rebooted. The board was assigned the correct number and programmed properly.

[image]

2. I found the pic32prog program won't program ports greater than 10. 

3. There is a registry clean that works. 
http://superuser.com/questions/408976/how-do-i-clean-up-com-ports-in-use
[image]

ricklon, 2015-07-30:
Thanks again. I'll configure it so the Windows version will have that prefix added.

ricklon, 2015-07-30:
I might have to copy a windows version of platforms.txt. Just to get the formatting proper.

ricklon, 2015-08-07:
The fix is now in the pic32prog program and is confirmed to work.

closed 2015-08-07 by ricklon