chipKIT32/chipKIT-core #26: Should drivers be included with the core download?

rianders, 2015-07-28:
Should drivers be included with the core download?

ricklon, 2015-07-29:
Confirmed. This should be done. A new directory called drivers will be added.

ricklon, 2015-07-31:
This should just be for Windows builds.

ricklon, 2015-09-23:
This is definitely more complicated when using the package json file.  The files are not easily accessible from the package directory.

JacobChrist, 2015-12-17:
I agree, these are missing, and I was about to add this issue.

EmbeddedMan, 2016-01-06:
Rick, I think that we should create a 'drivers' .zip file (just for windows) and include it as an additional 'tool' dependency in the .json file. We could then upload that zip file into the release for chipKIT-core. I think this is something I can do.

However, you'll also need to edit the build.xml file to include the drivers.zip file in the normal Windows .zip file for the whole package (for manual installs). 

Does this sound like the right thing to do?

ricklon, 2016-01-06:
That sounds great.  I can get crackin' on that.

--Rick

EmbeddedMan, 2016-01-07:
So I've created a drivers.zip file that I'm temporarily hosting on my EmbeddedMan GitHub repo release. We'll add that file to the chipKIT-core release v1.1.0 binary file list. However, how are you going to incorporate this drivers directory into the 'manual copy' chipKIT-core distribution Rick?

ricklon, 2016-01-07:
> Assigned #26 <https://github.com/chipKIT32/chipKIT-core/issues/26> to
> @ricklon <https://github.com/ricklon>.

Can you sign the file too? I can do a local signing if needed. The plan was
to pull it into the ./tmp directory. Then when a build is called bundle it
into the distro for manual. I've not worked out the separate download in
JSON yet.

EmbeddedMan, 2016-01-07:
Rick,

Hmm. Each of the two drivers included in the .zip file is already signed.
Why does the enclosing zip file also need to be signed? (I'm not even
really sure how to do that for just a zip file honestly.)

I've already worked out the separate download in the json file - that part
is done and working great. (See the EmbeddedMan json)

One of my questions is where should the build.xml process pull the windows
driver .zip file from? Right now it's in my EmbeddedMan/chipKIT-core
release, but I don't think that's a good place to put it long term. Maybe
it needs its own little separate repo?

*Brian

ricklon, 2016-01-08:
It should really be in it's own repo so that changes can be tracked and it would get pulled to the tmp directory.  Then will windows build occurs it would be retrieved from the release directory moved to the drivers folder.

Being in it's own release directy with a signature should allow it to be downloaded into tools for windows via the package.json technic. I hope.

EmbeddedMan, 2016-01-08:
Rick,

I'll put it in it's own repo then. But it doesn't need a signature to be
downloaded by .json file.We just need to put the accurate file size and
SHA256 value of the file in the .json file. Both drivers are already
signed, so Windows doesn't need anything more than that.

*Brian

ricklon, 2016-01-29:
These drivers included now. I've confirmed that it works.

closed 2016-01-29 by ricklon