chipKIT32/chipKIT-core #31: WF32 and WiFIRE do not build Blink

KeithV, 2015-08-03:
Get errors when either the WF32 or WiFIRE builds the example blink sketch

ricklon, 2015-08-22:
Discovered two issues. 

1. Both boards include .c/.cpp files inside of Board_Data.c. This is compilation error. This needs to be done a different way. All .c and .cpp files in the variant folder is now compiled. THis is different than MPIDE. This will be added back in MPIDE.

2. build.ldscript is used to find the ldscripts. Currently, testing changing the compilation recipes to make a valid ld feature.

EmbeddedMan, 2015-08-22:
Rick,

Are you going to be making the changes to the Board_Data.c files? Or do you
want me or Keith to do it?

*Brian

ricklon, 2015-08-22:
I'd like you or Keith to make the changes. I'm chasing down the next set of errors, but I think they are related to the testing that eliminated the error includes.

MPIDE
```
/Applications/mpide.app/Contents/Resources/Java/hardware/pic32/compiler/pic32-tools/bin/pic32-g++  -Os  -Wl,--gc-sections  -mdebugger  -mno-peripheral-libs  -nostartfiles  -mprocessor=32MZ2048ECG100  -o  /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build611625737197987005.tmp/wf32ex2serial.cpp.elf  /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build611625737197987005.tmp/wf32ex2serial.cpp.o    /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build611625737197987005.tmp/core.a  -L/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build611625737197987005.tmp  -lm  -T  /Applications/mpide.app/Contents/Resources/Java/hardware/pic32/variants/WiFire/MZ-application-32MZ2048ECX.ld  -T/Applications/mpide.app/Contents/Resources/Java/hardware/pic32/cores/pic32/chipKIT-application-COMMON-MZ.ld
```

Arduino
```
/Users/test/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++ -Os -Wl,--gc-sections -mdebugger -mno-peripheral-libs -nostartfiles -mprocessor=32MZ2048ECG100 -o /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7405389819665848001.tmp/wf32ex2serial.cpp.elf /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7405389819665848001.tmp/wf32ex2serial.cpp.o /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7405389819665848001.tmp/Board_Data.c.o /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7405389819665848001.tmp/core.a -L/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7405389819665848001.tmp -lm -T /Users/test/Documents/Arduino/hardware/chipkit-core/pic32/variants/WiFire/MZ-application-32MZ2048ECX.ld -T /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/chipKIT-application-COMMON-MZ.ld
```

error output:
```
/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7405389819665848001.tmp/core.a(wiring_analog.c.o): In function `analogRead':
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/wiring_analog.c:216: undefined reference to `convertADC'
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/wiring_analog.c:216: undefined reference to `convertADC'
/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7405389819665848001.tmp/core.a(WSystem.c.o): In function `_configSystem':
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WSystem.c:755: undefined reference to `initADC'
collect2: ld returned 1 exit status
```

EmbeddedMan, 2015-08-22:
Rick,

OK, I'm happy to make those changes. I will remove the including of any .c
or .cpp files from within the Board_Defs.c files in all variants. Are we
also doing to do this to MPIDE, or only chipKIT-core?

*Brian

ricklon, 2015-08-22:
We are going to backport that feature to MPIDE. I've filed the issue with
some sample code.

-_Rick

ricklon, 2015-08-22:
Worked out fix with Keith. See issue #10

ricklon, 2015-09-03:
Confirmed fixed.

closed 2015-09-03 by ricklon

KeithV, 2015-09-03:
Thanks Rick!