chipKIT32/chipKIT-core #32: Unable to build HTTPServer on uC32

KeithV, 2015-08-03:
Can not find the HTTPServerConfig.h file when building the example HTTP Server (for the uC32, but probably all boards fail). Can't find the header file even though the header file is in the sketch directory. Seems like the sketch directory is not on the include path.

ricklon, 2015-08-04:
The sketch folder seems to be missing from the include path.

ricklon, 2015-08-07:
I've confirmed this filed the issue with Arduino. I'm preparing a pull request for the fix. We'll see how it goes.

EmbeddedMan, 2016-01-06:
With the latest (v1.0.1) test release of chipKIT-core, this appears to be fixed. The deWebIOServer example compiles without error for uC32 now. The sketch folder is now always added to the include path when compiling.

Keith, can you confirm that this issue is now completely fixed?

KeithV, 2016-01-06:
Hi Brian, I am a confused as to the exact thing you want me to test?? I this just the platform.txt change to put the sketch folder on the include? That has been fixed for a while; or was there a secondary issue too? My take on this is that you are asking me to test the drop/build on amazon?

 

KeithV

EmbeddedMan, 2016-01-06:
Keith,

If you're confident that the original issue, as submitted, is fixed, then
I'll close the issue out. I was just making sure we closed the loop and
that there wasn't anything I was missing.

*Brian

closed 2016-01-07 by EmbeddedMan