chipKIT32/chipKIT-core #42: Put linker scripts back in original folder

ricklon, 2015-08-22:
Put linker scripts back in original folder

EmbeddedMan, 2015-10-25:
Rick, can you confirm is this is complete or not? I'm not exactly sure what the context of this issue is, so I can't confirm that it's been completed properly.

ricklon, 2016-01-29:
I'll double check. The issue was that linker scripts were in the variant folder. Originally, they weren't being found in that location. Which meant they had to be included in the "core" folder. Now I think we are back to being able to use scripts in the variant folder. I will confirm.

EmbeddedMan, 2016-01-29:
I am pretty sure that Keith has worked this all out, and we've added the
proper paths to the compiler command line such that the linker script files
in the varient folders is working properly now. But it would be good for
you to confirm just to be sure.