chipKIT32/chipKIT-core #51: Back-port Ethernet from the upstream Arduino core

majenkotech, 2015-08-31:
Our Ethernet library is horrendously out of date. It has files that conflict with files in the core and its inheritance tree is completely out of alignment with how things are done nowadays.

We need to copy the Arduino Ethernet library in from upstream and re-add the chipKIT specific parts to it.

majenkotech, 2015-08-31:
To do this we will need to also implement Udp.h in the core (a simple copy should suffice since it's pretty much a pure virtual template class).  That's the easy bit.

We will also need to back-port the new style SPI library that has transaction support.  That's going to be harder.  Better would be to adapt Ethernet to use DSPI instead of SPI, or make DSPI into SPI and standardize the interface.

majenkotech, 2015-08-31:
Udp.h added as pr #52

majenkotech, 2015-08-31:
New versions of SPI and Ethernet are now in testing.

majenkotech, 2015-08-31:
SPI is now submitted as pr #53

ricklon, 2015-09-13:
Confirmed this is added.

closed 2015-09-13 by ricklon