chipKIT32/chipKIT-core #63: Create 'commit number' text file in root

EmbeddedMan, 2015-09-06:
So when you take the chipkit-core folder out of the .zip file and put it in your Arduino\hardware\ folder, you loose the commit number and date part of the filename (from the zip file). This means that it's really hard to look at a chipkit-core folder and figure out which commit, or even which date, it came from. If we could automatically create a .txt file in the chipkit-core folder with a name that reflected the date and commit number of the build, then it would be obvious which chipkit-core version is currently in your hardware folder.

ricklon, 2015-09-23:
version.properties should include that information. If it doesn't, it should.

EmbeddedMan, 2015-10-25:
Yes, the version.properties files now contains CHIPKITCOREVER=16778039. However, that doesn't contain what my original intent for this issue was : namely, to have the build date or last commit number that triggered the build available somehow to easily compare multiple chipKIT-core builds. 

Also, this version.properties file does NOT come with when the dist is built. In other words. if you copy chipKIT-core down into your Arduino/hardware/ folder, you don't have any idea what date or commit number that chipKIT-core came from.

I wonder - how hard would it be to also create an empty file in the root of chipKIT-core (that would travel with it when zipped up in a dist) who's name would be something like "version-<date>-<commit>.txt" or something like that? Much like how we name our final output binaries now?

EmbeddedMan, 2016-01-06:
OK, with the advent of chipkitbot (and the output files that include date and build number), as well as a working .json file that includes a version number for the core and the tools, I think the need for this issue has disappeared. I'm closing it.

closed 2016-01-06 by EmbeddedMan