chipKIT32/chipKIT-core #67: Everyboard in boards.txt overides default compiler.c.flags, and compiler.cpp.flags

ricklon, 2015-09-13:
Every board in boards.txt overrides default compiler.c.flags, and compiler.cpp.flags even when they might not need to.

EmbeddedMan, 2015-09-17:
What's the task to do here? To go through each compiler.cpp.flags assignment in every variant and remove it if it's the same as he master copy in boards.txt?

ricklon, 2015-09-17:
Correct. If the board relies in default settings or needs custom settings.

--Rick

EmbeddedMan, 2015-09-17:
So for an option like this, in an individual board.txt file, there's now
way to specify "keep the existing value from the global boards.txt file,
but add my own little customization"? That's too bad.

I think I can take care of this one too.

*Brian

ricklon, 2015-09-17:
I think there is an equivalent of append, but I need to confirm Arduino
implemented it.

-_Rick

EmbeddedMan, 2015-09-17:
Rick, I'm a little confused as to how this is supposed to work.

We currently have compiler.c.flags being set in
platform.txt
platforms.txt
boards.txt
and in each of the boards.txt files that is in each variant folder.

Which of these does Arduino IDE actually use?

*Brian

ricklon, 2015-09-17:
/pic32
platform.txt
boards.txt

Arduino doesn't currently look in the variants folder. They use the
singular and MPIDE uses the plural.

--Rick

EmbeddedMan, 2015-09-17:
Rick,

OK, that makes sense. The problem is that every board in boards.txt has a
different value for compiler.c.flags than the value in platform.txt.

So maybe the right thing to do is for me to update the value that's in
platform.txt, then remove all of the (now redundant) values from
boards.txt. Does that make sense?

*Brian

ricklon, 2015-09-17:
That makes the most sense. But just do a test before converting everything.

We might need to file an issue with Arduino that their code doesn't append.

--Rick

EmbeddedMan, 2015-09-17:
How is the append feature supposed to work? If I understand that, I can
test it.

*Brian

EmbeddedMan, 2015-10-25:
I believe this issue is now fixed. Please confirm.