chipKIT32/chipKIT-core #68: Dependency check on recompilation not performed

ricklon, 2015-09-13:
Dependency check on recompilation not performed this is because the dependency files were not generated. 

Add the following to platform.txt and it generates properly.
compiler.cpp.extra_flags=-MMD
compiler.c.extra_flags=-MMD

If this value is changed in boards.txt dependencies will not be generated for a specific board.

EmbeddedMan, 2015-09-15:
Rick, I'm unable to verify that your PR fixed this issue. For an empty sketch, I get the same compile time first time vs. repeated compiles with no changes. (This is with latest chipKIT-core built from Github repo contents).

Did you test this and confirm that it fixed the problem on your system?

ricklon, 2015-09-15:
I'm double checking now. What board did you test with?

--Rick

EmbeddedMan, 2015-09-15:
Fubarino Mini.

ricklon, 2015-09-15:
I tested in the latest Arduino 1.6.6 master environment.

It didn't work in the Arduino  1.6.5.  The flag is included in the
compilation for both. Something has to be wrong about the comparison.

We need to file a new issue that it's not working in Arduino 1.6.5.

--Rick

EmbeddedMan, 2015-09-15:
Rick,

I'm also texting with 1.6.6 (the latest from Arduino's Github as of last
night) and I'm still not seeing any speedup. When I turn on verbose output,
every file is compiled every time. This is under Windows obviously.

*Brian

ricklon, 2015-09-15:
I wrote some debug code. It might be helpful.

I did a ```git stash show -p stash@{0}```

I added one line. This added cpp.hex to the list of hex files checked. That
might be why mine worked. eek!

``` "{build.path}/sketch/{build.project_name}.with_bootloader.hex",

       "{build.path}/sketch/{build.project_name}.hex",

+      "{build.path}/sketch/{build.project_name}.cpp.hex",```

On Tue, Sep 15, 2015 at 7:04 PM, Brian Schmalz <notifications@github.com>
wrote:

> Rick,
>
> I'm also texting with 1.6.6 (the latest from Arduino's Github as of last
> night) and I'm still not seeing any speedup. When I turn on verbose output,
> every file is compiled every time. This is under Windows obviously.
>
> *Brian
>
>
>
>
> On Tue, Sep 15, 2015 at 5:59 PM, Ricklon <notifications@github.com> wrote:
>
> > I tested in the latest Arduino 1.6.6 master environment.
> >
> > It didn't work in the Arduino 1.6.5. The flag is included in the
> > compilation for both. Something has to be wrong about the comparison.
> >
> > We need to file a new issue that it's not working in Arduino 1.6.5.
> >
> > --Rick
> >
> > On Tue, Sep 15, 2015 at 6:45 PM, Brian Schmalz <notifications@github.com
> >
> > wrote:
> >
> > > Fubarino Mini.
> > >
> > > On Tue, Sep 15, 2015 at 5:44 PM, Ricklon <notifications@github.com>
> > wrote:
> > >
> > > > I'm double checking now. What board did you test with?
> > > >
> > > > --Rick
> > > >
> > > > On Tue, Sep 15, 2015 at 6:17 PM, Brian Schmalz <
> > notifications@github.com
> > > >
> > > > wrote:
> > > >
> > > > > Rick, I'm unable to verify that your PR fixed this issue. For an
> > empty
> > > > > sketch, I get the same compile time first time vs. repeated
> compiles
> > > with
> > > > > no changes. (This is with latest chipKIT-core built from Github
> repo
> > > > > contents).
> > > > >
> > > > > Did you test this and confirm that it fixed the problem on your
> > system?
> > > > >
> > > > > —
> > > > > Reply to this email directly or view it on GitHub
> > > > > <
> > > >
> > >
> >
> https://github.com/chipKIT32/chipKIT-core/issues/68#issuecomment-140562032
> > > > >
> > > > > .
> > > > >
> > > >
> > > >
> > > >
> > > > --
> > > > Co-founder
> > > > Fair Use Building and Research (FUBAR) Labs
> > > > http://fubarlabs.org
> > > >
> > > > —
> > > > Reply to this email directly or view it on GitHub
> > > > <
> > >
> >
> https://github.com/chipKIT32/chipKIT-core/issues/68#issuecomment-140569235
> > > >
> > > > .
> > >
> > > >
> > >
> > > —
> > > Reply to this email directly or view it on GitHub
> > > <
> >
> https://github.com/chipKIT32/chipKIT-core/issues/68#issuecomment-140569389
> > >
> > > .
> > >
> >
> >
> >
> > --
> > Co-founder
> > Fair Use Building and Research (FUBAR) Labs
> > http://fubarlabs.org
> >
> > —
> > Reply to this email directly or view it on GitHub
> > <
> https://github.com/chipKIT32/chipKIT-core/issues/68#issuecomment-140576007
> >
>
> > .
> >
>
> —
> Reply to this email directly or view it on GitHub
> <https://github.com/chipKIT32/chipKIT-core/issues/68#issuecomment-140576685>
> .
>

-- 
Co-founder
Fair Use Building and Research (FUBAR) Labs
http://fubarlabs.org

ricklon, 2015-09-15:
One quick test would be to change the name of the hex file to not need
cpp.hex. That would match the Arduino default.

I saw that -MDD was included in the compile lines. That's the magic sauce.

EmbeddedMan, 2015-09-16:
Rick,

I don't see how changing the name of the hex file will have any effect
because the name *.cpp.hex isn't used until avr-objcopy is called at the
very end of the compile process. It doesn't have any effect on any of the
steps before that, which are the ones that should get skipped.

Could you post the complete output on your system when a full recompile is
done, and when the recompile is skipped?

*Brian

EmbeddedMan, 2015-09-16:
Rick, I found the problem. You committed a new platform.txt that has "-MDD"
and it should be "-MMD". I'll fix it right now and issue a PR.

WIth the correct spelling of the option, it does work properly.

It takes a 10 second build down to about 4 seconds for me. Not bad. As good
as Arduino anyway.

*Brian

ricklon, 2015-09-16:
Confirmed fixed!

Thank You!

--Rick

EmbeddedMan, 2015-10-25:
I have also confirmed that this issue is working as well as it's going to. (i.e. fixed.)

closed 2015-10-25 by EmbeddedMan