chipKIT32/chipKIT-core #78: Can't upload to UNO32 from core

EmbeddedMan, 2015-09-23:
MPIDE works just fine, so I know the board and the drivers and everything are just fine. And from Arduino IDE/core I'm able to upload to uc32 just fine. But there's something wrong with the current core's support for UNO32, at least for me. I'm compiling using the latest from the chipKIT-core repo, with Arduino 1.6.5, under Windows 8.1. Here is the error message I get:

Sketch uses 8,644 bytes (6%) of program storage space. Maximum is 126,976 bytes.
Global variables use 3,044 bytes of dynamic memory.
C:\Users\Brian\Documents\Arduino\hardware\chipkit-core/pic32/tools/bin/pic32prog.exe -d COM6 -b 115200 C:\Users\Brian\AppData\Local\Temp\buildcdf13c25b8169431d19b90675ce3d96d.tmp/sketch_sep23a.ino.hex 
Programmer for Microchip PIC32 microcontrollers, Version 2.0.157

No target found.
    Copyright: (C) 2011-2015 Serge Vakulenko

I tried re-installing the latest FTDI drivers (2.12.6.0) and I still have the same problem. My guess is that it has to do with pic32prog, but since this same exact setup works fine for uC32, I'm not sure what could be wrong.

ricklon, 2015-09-23:
If you copy that line to the command and paste it in what message do you
get?

--Rick

EmbeddedMan, 2015-09-23:
I get the exact same error (No target found) if I run that line from a command line. So this is an issue with PIC32PROG, not really chipKIT-core.

EmbeddedMan, 2015-10-01:
OK, I've found the problem. In the official Atmel STK500 protocol document (http://www.atmel.com/Images/doc2591.pdf) it lists that there are two acceptable responses to the CMD_SIGN_ON command: "STK500_2" or "AVRISP_2". And currently pic32prog is NOT looking for "AVRISP_2". And guess what response an UNO32 with Digilent bootloader uses? Yes, "AVRISP_2". No idea why - all other boards appear to respond with "STK500_2". 

So the solution is to fix this in pic32prog. I'll try to create  PR for it, and get a new binary inserted into chipKIT-core asap.

EmbeddedMan, 2015-10-02:
I have added "AVRISP_2" as a valid response string in pic32prog, and send a PR to the pic32prog project with this change. I have tested the change with a locally built pic32prog.exe and it works perfectly, allowing UNO32 boards to work properly under chipKIT-core and Arduino IDE with pic32prog. After pic32prog releases a new version with this fix and we get it into chipKIT-core, we can close this issue.

EmbeddedMan, 2015-10-03:
And pic32prog has now been updated. Whenever the next core build is done, it will pull in the latest pic32prog executable which has this fix and will thus work with UNO32s.

bkgoodman, 2015-10-13:
I am having the same problem with a Cerebot MX3CK (PIC-based). I am using the latest Chipkit Core for Arduino (2015-09-03). I'm assuming the fix isn't in that build. Any idea when/where to get a build for Arduino (Windows) with this fix?

EmbeddedMan, 2015-10-13:
bkgoodman,

I want to make sure that this is the problem you are seeing. Can you
download the pic32prog.exe from here
https://github.com/sergev/pic32prog
and replace your pic32prog with it, then restart Arduino and see if your
issue goes away?
Your pic32prog will be located at
C:\Users\Brian\Documents\Arduino\hardware\chipkit-core\pic32\tools\bin

If that works, let me know, and we'll get a test build with this change in
for you.

*Brian

EmbeddedMan, 2015-10-25:
This issue should be fixed, as long as chipkitbot is always doing an "ant clean-all" before it builds, as this will force the grabbing of the very latest pic32prog executable to be included in chipKIT-core. Can somebody confirm that this is the case? I believe that we need pic32prog version 2.0.183 or greater in our chipKIT-core zip files.

Revilones, 2015-10-26:
I am getting farther along but getting stuck on upload, below is example output. I am using a uC32 rev b, with what I assume is recent firmware. Also I have been able to successfully program my uC32 via MPIDE.

Sketch uses 10,536 bytes (2%) of program storage space. Maximum is 520,192 bytes.
Global variables use 3,056 bytes of dynamic memory.
Programmer for Microchip PIC32 microcontrollers, Version 2.0.180
    Copyright: (C) 2011-2015 Serge Vakulenko
      Adapter: STK500v2 Bootloader
 Program area: 1d000000-1d1fffff
    Processor: Bootloader
 Flash memory: 2048 kbytes
  Boot memory: 80 kbytes
         Data: 10528 bytes
        Erase: Cannot erase chip.

EmbeddedMan, 2015-10-26:
Revilones, it looks like you're using pic32prog rev 2.0.180. I wonder if using the latest pic32prog (2.0.183) would help solve the problem you're seeing. It may not, but I think it's worth a shot. Are you able to pull the right executable down from here https://github.com/sergev/pic32prog and put it in your chipkit-core/pic32/tools/bin/ directory and give it another try?

Revilones, 2015-10-27:
I checked out the pic32prog you mentioned and it's the same version 

bash-3.2$ ./pic32prog -V Programmer for Microchip PIC32 microcontrollers, Version 2.0.180

Also if it helps I am using OSX

I am more than happy to do some debugging. Do you have any pointers of where I should start looking first?

EmbeddedMan, 2015-10-27:
I need to look into this some more - the Windows version is 2.0.183. I'm
not sure why the Linux version isn't the same. Maybe it doesn't make any
difference. Hmm.

If I can find any debugging ideas for you, I will pass them on.

jpswensen, 2015-10-28:
I have a solution for this one. I had to go program the latest bootloader to get the "Cannot erase chip." error to go away. I got the bootloader from
https://github.com/chipKIT32/PIC32-avrdude-bootloader
and installed it using their sketch installer.

I am on a uc32 right now, but will test on my uno32 I have at home later tonight. If this works, you can problem close out this and Issue #92

EmbeddedMan, 2015-10-28:
John,

If this problem truly was because of a bad uC32 bootloader, Digilent should
be made aware of it so they can see if others have the same problem. I'm
cc'ing Keith Vogel from Digilent on this issue so he can take a look. It's
very interesting to me that the bootloader from the chipKIT GitHub solved
your problem - and I'm super pumped that you were able to use the update
sketch there to put the new bootloader on. (Keith wrote all those
sketches.)

*Brian

EmbeddedMan, 2015-10-28:
Hi Brian / John,

To be honest I didn’t even know the bootloader update sketches were in the bootloader repro…. Good work there Brian!

Looks like the one out there for the uC32 is 2 years old; but I am glad it worked. I did make a newer one and it is posted on our product page at:

http://www.digilentinc.com/Data/Products/CHIPKIT-UC32/chipKIT_uC32_BL_Update.zip

There are several issues we are fighting here; and most of this we (average users) never think about because they are typically handled for us…

1.       The linker scripts place things where they belong. If someone fools with the linker scripts things will definitely go wrong! Likewise, using the wrong linker script is a problem as well. If we are pointing to the wrong linker scripts in the IDE, we need to fix that.

2.       Get the compiler options correct… yep if we get that wrong we can start placing things where they can’t be. The directive was clearly telling the compiler to put it exactly where it was placed in the order of the declarations in the code (an implied pack) and that caused misalignment. Thanks Brian for fixing that!

3.       Sometimes, if the wrong compiler option or linker script is used, overlapping code can be specified in the hex file, the bootloader will dutifully try to do that, but since flash can only be written once, the result of overwriting flash is garbage and that will fail the verification in either AVRDude or pic32prog.

4.       And your last problem was a really old bootloader on the uC32. This was a procedural error at Digilent allowing basically the “original” bootloader to be loaded on the uC23. That bootloader works fine with MPIDE/AVRDude on Windows, Mac and Linux can offers some issues. While we have updated the bootloaders twice over the years, for whatever reason that did not make its way to manufacturing. The bootloader update sketch fixes that problem. Worst case, you will have to program the update sketch from MPIDE/Windows; that should always work. When I built those update sketches, I made sure the .HEX files were built correctly, thus you will not get alignment errors or out of memory errors. But just because you have the latest bootloader doesn’t mean you can’t use bad compiler options or wrong linker scripts when building your sketces; that will always create problem no matter what bootloader you use.

I am glad you resolved your problem.
KeithV

https://github.com/chipKIT32/PIC32-avrdude-bootloader

xxxajk, 2015-11-16:
This is a bug in pic32prog.
I have fixed it for non-windows, and notified the author of the program about the bug, and included a pull request.
https://github.com/sergev/pic32prog/pull/24

Nate42, 2016-01-06:
This issue now seems to effect uc32 as well. Updating to latest bootloader did not help. Works fine with MPIDE, Arduino IDE + chipkit-core says "No target found".

EmbeddedMan, 2016-01-06:
Nate42, what OS are you running your tests on?

*Brian

xxxajk, 2016-01-06:
I had same issue, but fixed the uploader.
https://github.com/xxxajk/pic32prog

EmbeddedMan, 2016-01-06:
Andrew, what's the status of your PR for pic32prog? Has it been merged and
tested yet? If not, do you know what the holdup is?

Nate42, 2016-01-06:
I'm on Windows 7 64 bit at the moment. I could try it with Window 10 from
home later today if that helps.

Nate

jpswensen, 2016-01-06:
My problem on both Uno33 and uc32 were solved with a boot loader update and building chip kit core from github. I think this ensures the chip kit problems are solved, the boot loader problems are solved, and the picprog32 problems are solved.

EmbeddedMan, 2016-01-06:
Nate, - no need to test on Win10. As far as this is concerned, I think it
will work identically to Win7.

Which version of chipKIT-core did you notice this problem on?

If you want, you can try the latest build from here and see if it solves
your problem :
http://chipkitjenkins.wayneandlayne.com/job/chipKIT-core/lastBuild/

*Brian

Nate42, 2016-01-06:
I'm running build 48 from Dec 31, so slightly out of date. I can try the
latest and greatest and let you know. build 48 did work fine with my
Wi-Fire board btw. Thanks for your help I'll let you know if that one works.

Nate

Nate42, 2016-01-06:
Just now got a chance to try the latest dev build, and it works like a
charm. So I guess this is fixed, thanks guys and sorry for the false alarm.

Nate

EmbeddedMan, 2016-01-06:
Nate- no problem at all. Thanks for taking the time to test it out.

*Brian

EmbeddedMan, 2016-01-06:
This issue appears to be solved by using the latest pic32prog version.

closed 2016-01-06 by EmbeddedMan