chipKIT32/chipKIT-core #81: Add build.path to include list

EmbeddedMan, 2015-09-30:
The build.path (the temporary folder where everything gets copied to before the compile happens) should be added to the list of include paths on the compiler command line, so that any .h files that are in the sketch folder can be included in our sketches and libraries.

EmbeddedMan, 2015-10-25:
I believe this issue has been fixed. Please confirm.

EmbeddedMan, 2016-01-07:
Keith, I believe you are the one who found this issue in the first place. Can you confirm here that this is fixed to your satisfaction?

EmbeddedMan, 2016-01-14:
Keith, did you have a chance to confirm that this issue has been addressed to your satisfaction?

KeithV, 2016-01-15:
This works,but you might have a redundant include to the build directory that is no longer used. You only need the build\sketch directory. If that is fixed you can close.

EmbeddedMan, 2016-01-26:
Keith - thanks! I have fixed the redundant build directory as well.

closed 2016-01-26 by EmbeddedMan