chipKIT32/chipKIT-core #86: Can dist zips have a branch name in them if they aren't on master?

adamwolf, 2015-10-02:
Hi Rick (and others),

My ant-fu is extremely weak.

1) How hard would it be to get it so that dist zipfiles have the branch name in them, if the branch name isn't master?

2) Does anyone else think this is a good idea?

Thanks!

EmbeddedMan, 2015-10-02:
Yeah, not only a good idea, a pretty critical one.

adamwolf, 2015-10-05:
Do we mind if it always has the branch name, or do we want it to only have the branch name in the zip if it isn't master?

EmbeddedMan, 2015-10-05:
I'm not opposed to every build including the branch name. I like the
consistency.

I do have to admit that I don't like the current build naming scheme much
because it does not include the piece of information that I need most of
all (other than the platform) which is the date on which it was built.  I
have to get used to using the build number (16778041 in the filename below)
to know which one of several builds is the latest. We were used to doing it
with dates in the filenames, but that has changed.

Example: chipkit-core-windows.16778041.v1.0.0-21-g1c4b889.zip

*Brian

EmbeddedMan, 2015-10-25:
Adam - what is the current status of this issue?
Rick - is there any reason to always put the branch name in the build (no matter if it's a real branch or master)?

chipkitbot, 2015-10-25:
I was holding off on this until I knew if the weird zip issue was going to
come back.  That's been fine for a while, and was only an issue on that one
branch. I can get tot his early this week.

EmbeddedMan, 2015-10-25:
Sweet. Thanks chipkitbot! You rock.

adamwolf, 2015-10-29:
@EmbeddedMan Can you pull from my personal branch called branchname_in_tarballs and see what your dist zips show up as when you build on your desktop?  It works on Jenkins, so now I need to make sure it works on a regular system.

EmbeddedMan, 2015-10-29:
I can confirm that it works for me locally under Windows. Pull the trigger!

*Brian

closed 2015-10-30 by adamwolf