chipKIT32/chipKIT-core #92: "odd address" error when trying to program

jpswensen, 2015-10-16:
Platform: OSX
Version: 20151016 from GitHub
ArduinoIDE version: 1.6.5

Whenever I try to load to the chipkit uc32 board I get the following error:
Programmer for Microchip PIC32 microcontrollers, Version 2.0.180
    Copyright: (C) 2011-2015 Serge Vakulenko
/var/folders/wb/0y2_jh0j2xd52ll_zpb1p1y40000gn/T/build4751426172754605784.tmp/Blink2.cpp.hex: odd address

This seems to be an error given from pic32prog, but it looks like it is because of how the binary was generated.

Any suggestions?

jpswensen, 2015-10-16:
As an added investigation, I compiled the same code under both MPIDE and the ArduinoIDE+chipKIT-core and verified that the they do, in fact generate different addresses in the .hex file. MPIDE doesn't have odd addresses and ArduinoIDE+chipKIT-core does. I then tried to look at the linker output for both environments and they looked essentially identical.

1) Just for kicks, I compared whether the chipKIT-application-32MX340F512.ld and the chipKIT-application-COMMON.ld used in the linker step are the same for both platforms. They are not. So, I copied these .ld files from the MPIDE distribution to the appropriate chipKIT-core folder. This fixed almost all the odd address problems (early in the .hex file), but there are still about 60 lines at the end of the .hex file with odd addresses.
2) Also, I can't see the parameters used for the pic32-objcopy and pic32-bin2hex steps of the MPIDE build to see how they differ from MPIDE

EmbeddedMan, 2015-10-18:
I am getting the same error with the UBW32-460 board (the only one I've tried) - it may be affecting all boards with this latest build of chipKIT-core. This is on Windows.

EmbeddedMan, 2015-10-18:
Removing the -fno-toplevel-reorder (in both places) from the platforms.txt file fixes this particular problem. But I don't know why.

EmbeddedMan, 2015-10-18:
Besides a lot of other code differences, these lines are added to the end of the hex file when -fno-toplevel-reorder is used during the compile. Note the odd addresses:

:020000040000fa
:020000041d00dd
:01331400694f
:020000040000fa
:020000041d00dd
:01331500684f
:020000040000fa
:020000041d00dd
:01331600674f
:020000040000fa
:020000041d00dd
:01331700664f
:020000040000fa
:020000041d00dd
:01331800694b
:020000040000fa
:020000041d00dd
:01331900684b
:020000040000fa
:020000041d00dd
:01331a00674b
:020000040000fa
:020000041d00dd
:01331b00664b
:020000040000fa
:020000041d00dd
:01331c006947
:020000040000fa
:020000041d00dd
:01331d006847
:020000040000fa
:020000041d00dd
:01331e006747
:020000040000fa
:020000041d00dd
:01331f006647
:020000040000fa
:020000041d00dd
:013320006943
:020000040000fa
:020000041d00dd
:013321006843
:020000040000fa
:020000041d00dd
:013322006743
:020000040000fa
:020000041d00dd
:013323006643
:020000040000fa
:020000041d00dd
:01332400693f
:020000040000fa
:020000041d00dd
:01332500683f
:020000040000fa
:020000041d00dd
:01332600673f
:020000040000fa
:020000041d00dd
:01332700663f
:020000040000fa
:020000041d00dd
:01332800693b
:020000040000fa
:020000041d00dd
:01332900683b
:020000040000fa
:020000041d00dd
:01332a00673b
:020000040000fa
:020000041d00dd
:01332b00663b

EmbeddedMan, 2015-10-18:
OK, I found the fix for this issue. Turns out I put the wrong compiler command line option in the platform.txt file. Instead of "-fno-toplevel-reorder" it should be "-ftoplevel-reorder" in two places. I will fix.

jpswensen, 2015-10-19:
Thanks for getting on this quick. I appreciate it. We have been using both Arduinos and chipKits in the Mechatronics class I teach and I wanted to have a single IDE for them regardless of which board they are using (some also use various Arduino derivatives for senior project and other course).

EmbeddedMan, 2015-10-19:
John, - no problem. Try out one of the new builds, and make sure that it
solves the problem for you. Let us know if it doesn't.

Arm:
https://chipkit.s3.amazonaws.com/chipkit-core/builds/arm/chipkit-core-arm.16778039.v1.0.0-27-gdfa0309.zip
Linux32:
https://chipkit.s3.amazonaws.com/chipkit-core/builds/linux32/chipkit-core-linux32.16778039.v1.0.0-27-gdfa0309.zip
Linux64:
https://chipkit.s3.amazonaws.com/chipkit-core/builds/linux64/chipkit-core-linux64.16778039.v1.0.0-27-gdfa0309.zip
MacOS:
https://chipkit.s3.amazonaws.com/chipkit-core/builds/macosx/chipkit-core-macosx.16778039.v1.0.0-27-gdfa0309.zip
Windows:
https://chipkit.s3.amazonaws.com/chipkit-core/builds/win/chipkit-core-windows.16778039.v1.0.0-27-gdfa0309.zip

EmbeddedMan, 2015-10-25:
This issue should be fixed. Please confirm.

jpswensen, 2015-10-27:
It didn't fix it. It gets further now, however it now gets to the point where the picprog32 says "Cannot erase chip" and then it is stuck in the "Uploading state forever". The lights are blinking like it is in the programming mode, but it never finishes

```
Sketch uses 10,520 bytes (2%) of program storage space. Maximum is 520,192 bytes.
Global variables use 3,052 bytes of dynamic memory.
C:\Users\mme-user\Desktop\arduino-1.6.5-r5\hardware\chipkit-core/pic32/tools/bin/pic32prog.exe -d COM8 -b 115200 C:\Users\mme-user\AppData\Local\Temp\build4449191092106776569.tmp/Blink.cpp.hex 
Programmer for Microchip PIC32 microcontrollers, Version 2.0.183
    Copyright: (C) 2011-2015 Serge Vakulenko
      Adapter: STK500v2 Bootloader
 Program area: 1d000000-1d1fffff
    Processor: Bootloader
 Flash memory: 2048 kbytes
  Boot memory: 80 kbytes
         Data: 10512 bytes
        Erase: Cannot erase chip.
```

jpswensen, 2015-10-28:
We can probably close this issue as my new problem seems the same as issue #78

EmbeddedMan, 2015-10-28:
I agree. The original problem here with pic32prog has been fixed.

closed 2015-10-28 by EmbeddedMan