chipKIT32/chipKIT-core #102: Duplicate chipKIT32-MAX issue #591 in chipKIT-core

EmbeddedMan, 2015-10-26:
https://github.com/chipKIT32/chipKIT32-MAX/issues/591

This is passing the startup assembly file directly to the linker rather than as part of the archive.

EmbeddedMan, 2015-10-26:
So I gave it a shot, and ended up with this change in platform.txt:

recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mprocessor={build.mcu} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" **"{build.ldscript.path}/cpp-startup.S"** {object_files} "{build.path}/{archive_file}" -L{build.path} -lm  -T "{build.ldscript.path}/{ldscript}" -T "{build.core.path}/{ldcommon}"

But now, with just this one little change, I get the following when I compile:

C:\Users\Brian\Documents\Arduino\hardware\chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++ -Os -Wl,-M=C:\Users\Brian\AppData\Local\Temp\build1651404181962169181.tmp/sketch.map --gc-sections -mdebugger -mno-peripheral-libs -nostartfiles -mprocessor=32MX250F128D -o C:\Users\Brian\AppData\Local\Temp\build1651404181962169181.tmp/Blink.cpp.elf C:\Users\Brian\Documents\Arduino\hardware\chipkit-core\pic32\cores\pic32/cpp-startup.S C:\Users\Brian\AppData\Local\Temp\build1651404181962169181.tmp\Blink.cpp.o C:\Users\Brian\AppData\Local\Temp\build1651404181962169181.tmp\Board_Data.c.o C:\Users\Brian\AppData\Local\Temp\build1651404181962169181.tmp/core.a -LC:\Users\Brian\AppData\Local\Temp\build1651404181962169181.tmp -lm -T C:\Users\Brian\Documents\Arduino\hardware\chipkit-core\pic32\cores\pic32/chipKIT-application-32MX250F128.ld -T C:\Users\Brian\Documents\Arduino\hardware\chipkit-core\pic32\cores\pic32/chipKIT-application-COMMON.ld 

cc1.exe: error: unrecognized command line option "-fgc-sections"
Error compiling.

But this makes no sense, because I don't have "-fgc-sections" on the command line! What's going on here?

jasonkajita, 2015-10-26:
We need the --gc-sections option to be passed through to the linker. We can do this using "-Wl,--gc-sections".

EmbeddedMan, 2015-10-27:
Right, OK. But here's the link command line before the change (this
generates no errors)

C:\Users\Brian\Documents\Arduino\hardware\chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++
-Os
-Wl,-M=C:\Users\Brian\AppData\Local\Temp\build4013326053842459685.tmp/sketch.map
--gc-sections -mdebugger -mno-peripheral-libs -nostartfiles
-mprocessor=32MX250F128D -o
C:\Users\Brian\AppData\Local\Temp\build4013326053842459685.tmp/Blink.cpp.elf
C:\Users\Brian\AppData\Local\Temp\build4013326053842459685.tmp\Blink.cpp.o
C:\Users\Brian\AppData\Local\Temp\build4013326053842459685.tmp\Board_Data.c.o
C:\Users\Brian\AppData\Local\Temp\build4013326053842459685.tmp/core.a
-LC:\Users\Brian\AppData\Local\Temp\build4013326053842459685.tmp -lm -T
C:\Users\Brian\Documents\Arduino\hardware\chipkit-core\pic32\cores\pic32/chipKIT-application-32MX250F128.ld
-T
C:\Users\Brian\Documents\Arduino\hardware\chipkit-core\pic32\cores\pic32/chipKIT-application-COMMON.ld

Notice that it goes "-Wl,<mapfile> --gc-sections". So why does that work
then if it should be "-Wl,--gc-sections"?

jasonkajita, 2015-10-27:
It probably didn't cause an error before since the compiler wasn't getting called, only the linker. Passing the *.S file is causing the compiler to get called for preprocessing. The error is getting caught.

So while there wasn't an error message before, unused sections may not have been getting removed.

-

EmbeddedMan, 2015-10-27:
Oh, wow! Then I'm really glad I caught this. I'll re-work the command line
tonight and see if I can get it working right.

EmbeddedMan, 2015-10-29:
I fixed this, and merged in the change that fixes it. Moving the map file path out from between the -Wl, and the --gc-sections makes --gc-sections actually work! And our sketches just got way smaller! SuperWoot!

EmbeddedMan, 2016-01-06:
This issue is fixed.

closed 2016-01-06 by EmbeddedMan