chipKIT32/chipKIT-core #108: Update build.xml to pull in new v1.40 C compiler

EmbeddedMan, 2015-11-05:
During the build process, we should now pull in the v1.40 release of the Chipkit C compiler : https://github.com/chipKIT32/chipKIT-compiler-builds

ricklon, 2015-11-05:
> Assigned #108 <https://github.com/chipKIT32/chipKIT-core/issues/108> to
> @ricklon <https://github.com/ricklon>.

Thanks. I'll get on that.
--Rick

ricklon, 2015-11-14:
I can get it done today. The one thing would be great is for these files to have a checksum that comes with them. Currently, to make package.json I have do the following to create checksums:
```
<target name="checksum-pic32tools">
    <checksum algorithm="SHA-256" fileext=".asc" forceOverwrite="yes">
      <fileset dir="tmp">
        <include name="pic32-tools*.zip"/>
      </fileset>
    </checksum>
  </target>
```

EmbeddedMan, 2015-11-14:
Rick, I don't think that's unreasonable at all.  I'll poke Jason about it. For now, can we proceed with your above hack until the compiler comes with it's own checksum?

ricklon, 2015-11-14:
Yep. I'll proceed as normal. Working on adding it in now.

EmbeddedMan, 2016-01-06:
I have confirmed that chipKIT-core build.xml file is downloading pic32-tools from the v1.40rc2 release.

closed 2016-01-06 by EmbeddedMan