chipKIT32/chipKIT-core #110: Adjust chipKIT-core to work with Arduino 1.6.6 platform level changes

ricklon, 2015-11-06:
Adjust chipKIT-core to work with Arduino 1.6.6 platform level changes. Changes were made to the platform variables. I've not been able to find clear documentation. But regardless it's effecting the chipKIT-core inside Arduino 1.6.6.

Current error:
```
% "/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++"  -DF_CPU=80000000L -DARDUINO=10606 -DARDUINO_BOARD_FUBARINO_SD_ -DARDUINO_ARCH_PIC32 -MMD -D_USE_USB_FOR_SERIAL_ "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/sketch/blink1.ino.cpp"
/Users/rickanderson/Documents/Arduino/blink1/blink1.ino: In function 'void setup()':
/Users/rickanderson/Documents/Arduino/blink1/blink1.ino:9:21: error: 'OUTPUT' was not declared in this scope
/Users/rickanderson/Documents/Arduino/blink1/blink1.ino:9:27: error: 'pinMode' was not declared in this scope
/Users/rickanderson/Documents/Arduino/blink1/blink1.ino:10:21: error: 'INPUT' was not declared in this scope
/Users/rickanderson/Documents/Arduino/blink1/blink1.ino:12:3: error: 'Serial' was not declared in this scope
/Users/rickanderson/Documents/Arduino/blink1/blink1.ino: In function 'void loop()':
/Users/rickanderson/Documents/Arduino/blink1/blink1.ino:42:7: error: 'Serial' was not declared in this scope
```
This seems to be that it can't find the core or object files for some reason.

I saw a new platforms.txt file added. It seems to be at the /hardware folder for the Arduino AVR code. 

```
 cat platform.txt 
# ctags
# ------------------------------
tools.ctags.path={runtime.tools.ctags.path}
tools.ctags.cmd.path={path}/ctags
tools.ctags.pattern="{cmd.path}" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "{source_file}"

# additional entries
tools.avrdude.path={runtime.tools.avrdude.path}

preproc.includes.flags=-w -x c++ -M -MG -MP
recipe.preproc.includes="{compiler.path}{compiler.cpp.cmd}" {preproc.includes.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}"

preproc.macros.flags=-w -x c++ -E -CC
preproc.macros.compatibility_flags={build.mbed_api_include} {build.nRF51822_api_include} {build.ble_api_include} {compiler.libsam.c.flags} {compiler.arm.cmsis.path} {build.variant_system_include}
recipe.preproc.macros="{compiler.path}{compiler.cpp.cmd}" {compiler.cpreprocessor.flags} {compiler.cpp.flags} {preproc.macros.flags} -DF_CPU={build.f_cpu} -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {preproc.macros.compatibility_flags} {includes} "{source_file}"

```

ricklon, 2015-11-07:
I think the problem is in the new feature of finding the dependencies. The process doesn't stop on error it keeps on going. So the following command fails:
```
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++"  -DF_CPU=48000000L -DARDUINO=10606 -DARDUINO_BOARD_FUBARINO_MINI_ -DARDUINO_ARCH_PIC32 -G1024 -D_USE_USB_FOR_SERIAL_ "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_Mini" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build65d3870c8bf55ad954f468e40e2b150e.tmp/sketch/FubarinoMiniLightAnimation.ino.cpp"
/Users/rickanderson/Documents/Arduino/FubarinoMiniLightAnimation/FubarinoMiniLightAnimation.ino:1:20: fatal error: WS2812.h: No such file or directory
compilation terminated.
```
But the compile process continues. So in the midst of error messages it's hard to find the trouble.

--Rick

rianders, 2015-11-14:
Since I can't see what's happening when I run arduino-builder from inside Arduino 1.6.6. I've started to number each configuration line and check for what number the errors occur on. SO. I've got most of the prepoc succeed in it's action. I sitll don't know how well it's working. But here's current configuration:

```

preproc.includes.flags=-w -x c++ -M -MG -MP
recipe.preproc.includes="{compiler.path}{compiler.cpp.cmd}"  {compiler.cpp.flags} {preproc.includes.flags}  -mprocessor={build.mcu} -DF_CPU={build.f_cpu}  -DARDUINO={runtime.ide.version} -D_{build.board} {compiler.define} "{compiler.cpp.extra_flags}" {build.extra_flags} {includes} "{source_file}"
#-o "{object_file}"
preproc.macros.flags=-w -x c++ -E -CC
preproc.macros.compatibility_flags=222
recipe.preproc.macros="{compiler.path}{compiler.cpp.cmd}"  {compiler.cpp.flags} {preproc.macros.flags} -mprocessor={build.mcu} -DF_CPU={build.f_cpu}  -DARDUINO={runtime.ide.version} -D_{build.board} {compiler.define} "{compiler.cpp.extra_flags}" {build.extra_flags} {includes} "{source_file}"
```

So now I need to move on to ctags. Does pic32 have a command equivalent to ctags?

Curren config:
```
tools.ctags.path={runtime.tools.ctags.path} 444
tools.ctags.cmd.path={path}/ctags 555
tools.ctags.pattern="{cmd.path}"  -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "{source_file}"

```

rianders, 2015-11-14:
I just updated to use the Arduino ctags command. The following occured:
```
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino2/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/preproc/ctags_target.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino2/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/preproc/ctags_target.cpp"
```

With this configuration:
```
tools.ctags.path={runtime.tools.ctags.path}
tools.ctags.cmd.path={path}/ctags
tools.ctags.pattern="{cmd.path}" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns "{source_file}"
```

rianders, 2015-11-14:
With this configuration a core.a is created but in the wrong place:
```
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++" -Os -Wl,--gc-sections -mdebugger -mno-peripheral-libs -nostartfiles -M=/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/sketch.map -mprocessor=32MX250F128D  -o "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/blink1.ino.elf" "/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/cpp-startup.S" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/sketch/blink1.ino.cpp.o" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/core/core.a" -L/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp -lm  -T "/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/chipKIT-application-32MX250F128.ld" -T "/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/chipKIT-application-COMMON.ld"
pic32-g++: /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildab83a842c2270138901fac8ecd68bff1.tmp/core/core.a: No such file or directory
exit status 1
Error compiling.
```

ricklon, 2015-11-14:
Ok. I'm closing in on the issue. 
```
"{build.path}/{archive_file}"
```
Needs to be replaced with:
```
"{archive_file_path}" 
```

Final ar pattern:
```
recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}"  {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}"  "{object_file}"
```

EmbeddedMan, 2015-11-14:
Rick,

Nice work! So is core now fully functional in 1.6.6?

ricklon, 2015-11-14:
Please test and see. This works with blink. It's supposed to work with all our libraries to but it needs some testing.

EmbeddedMan, 2015-11-14:
Rick you'll need to do a pull request so I can put all of your changes in
and test them.

EmbeddedMan, 2015-12-21:
I believe this is now fixed in the latest chipKIT-core build (http://chipkitjenkins.wayneandlayne.com/job/chipKIT-core%20PR%20Builder/65/). I can't reproduce this error with the changes to platform.txt that have been made.

closed 2016-01-06 by EmbeddedMan