chipKIT32/chipKIT-core #111: SERIAL_PORT_MONITOR definition missing

xxxajk, 2015-11-12:


EmbeddedMan, 2015-11-14:
These will have to go in the variant file for each board we support, as they will be different for different boards. Also, the new Harmony USB integration will re-jigger these around even on the same board.

xxxajk, 2015-11-16:
Since I only have the UNO32, I only made the recommendation, and no patch, but yes, that would be an issue, however I don't think all is lost.
The general idea used by the Arduino team is to use something along the lines of some sort of sane defaults, which is better than no definition whatsoever. There is also always the ability to add extra menu items in order to control the outcome and get what you want.

What I have done in the upcoming UHS 3.0 is to implement all code into inline coded headers, and allow the user to define options in the sketch. For example, see line 9 at https://github.com/felis/UHS30/blob/master/libraries/USB_HOST_SHIELD/examples/TWO_HOSTS_BULK/TWO_HOSTS_BULK.ino
Notice that while serial debug output is configured within the library, that there is a way to override it. In order for me to support the UNO32, and other microchip based boards out-of-the-box, at least some sort of output default is needed. It does not have to be the "Serial" class, even a USB serial class is fine, like on the DUE and Zero.

While the following idea won't work well for something that is a core, it could work out if Serial were to be moved to an inline library, but then you do have the issue of having to include it in the sketch, which unfortunately is not how the Arduino team originally did it. They assumed some things would always be there, even if you did not use them -- which can bloat the binary. Unfortunately they also automatically have Arduino.h included too before anything else, and do other evil rewrites of the sketch, such as prototyping the contents. Fortunately there are some other ways around these problems, but I am already beyond the scope of the fix request already.

xxxajk, 2016-02-13:
So still no fix for this, or #114? Come on, it's not like it is rocket science.

JacobChrist, 2016-07-07:
Pull requests welcome.

xxxajk, 2016-07-07:
I've been quite busy, I'll try to get this in this morning... I usually code/work at night 8-)

xxxajk, 2016-08-11:
Finally got to this.
https://github.com/chipKIT32/chipKIT-core/pull/247

xxxajk, 2018-12-31:
long long ago merged, closing.

closed 2018-12-31 by xxxajk