chipKIT32/chipKIT-core #112: 1.6.6 Compile error for WiFIRE create bogus .d file in sketch folder

ricklon, 2015-11-15:
1.6.6 Compile error for WiFIRE create bogus .d file in sketch folder 

```

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/rickanderson/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/rickanderson/Documents/Arduino/libraries" -fqbn=chipkit-core:pic32:fubarino_sd -ide-version=10606 -build-path "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7db2176c9dece65e48aa570237950620.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/rickanderson/projects/ricklon/circuitsunday/week1/blink/blink.ino"
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/Arduino.app/Contents/Java/hardware" -hardware "/Users/rickanderson/Documents/Arduino/hardware" -tools "/Applications/Arduino.app/Contents/Java/tools-builder" -tools "/Applications/Arduino.app/Contents/Java/hardware/tools/avr" -built-in-libraries "/Applications/Arduino.app/Contents/Java/libraries" -libraries "/Users/rickanderson/Documents/Arduino/libraries" -fqbn=chipkit-core:pic32:fubarino_sd -ide-version=10606 -build-path "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7db2176c9dece65e48aa570237950620.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Users/rickanderson/projects/ricklon/circuitsunday/week1/blink/blink.ino"
WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'
WARNING: Spurious .git folder in 'ckws2812' library
"/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/pic32-g++"  -O2 -c -mno-smart-io -w -fno-exceptions -ffunction-sections -fdata-sections -g -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -w -x c++ -M -MG -MP  -mprocessor=32MX795F512H -DF_CPU=80000000L  -DARDUINO=10606 -D_BOARD_FUBARINO_SD_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino "-MMD" -D_USE_USB_FOR_SERIAL_ "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32" "-I/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD" "/var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/build7db2176c9dece65e48aa570237950620.tmp/sketch/blink.ino.cpp"
<built-in>:0:0: fatal error: opening dependency file blink.ino.d: Permission denied
compilation terminated.
exit status 1
Error compiling.
````

EmbeddedMan, 2015-12-21:
Is this bug still accurate? Can it be reproduced with the latest chipKIT-core build (http://chipkitjenkins.wayneandlayne.com/job/chipKIT-core%20PR%20Builder/65/) and the release of Arduino 1.6.7?

EmbeddedMan, 2016-01-06:
I believe this issue is fixed in the latest chipKIT-core v1.0.1 test release. I am unable to reproduce it in that environment.

closed 2016-01-06 by EmbeddedMan