chipKIT32/chipKIT-core #117: Mac OS X chipKIT compiler error

rianders, 2015-12-16:
The previous compiler works with the Arduino Nightly, and current platforms.txt file. But the new compiler is failing with the following messages:

Fubarino Mini
```
WARNING: Category '' in library OneWire is not valid. Setting to 'Uncategorized'
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.c:85:0:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_Mini/Board_Data.c:274:5: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     _PPS_OUT(_PPS_RPB10),   //  31  RB10
     ^
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_Mini/Board_Data.c:275:5: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     _PPS_OUT(_PPS_RPB11),   //  32  RB11
     ^
cc1: fatal error: opening output file chipKIT-application-32MX250F128.ld.00: Permission denied
compilation terminated.
cc1: fatal error: opening output file chipKIT-application-32MX250F128.ld.01: Permission denied
compilation terminated.
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/bin/gcc/pic32mx/4.8.3/../../../../bin/pic32-ld: cannot open linker script file chipKIT-application-32MX250F128.ld.00: No such file or directory
collect2: error: ld returned 255 exit status
exit status 255
Error compiling.

```

Fubaino SD
```
n file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:263:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from sketch/sketch_dec16a.ino.cpp:1:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD/Board_Defs.h:229:0: warning: "portRegisters" redefined [enabled by default]
 #define portRegisters(P) ( port_to_tris_PGM[P] )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from sketch/sketch_dec16a.ino.cpp:1:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:233:0: note: this is the location of the previous definition
 #define portRegisters(P) ((p32_ioport *)(port_to_tris_PGM[P]))
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:263:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from sketch/sketch_dec16a.ino.cpp:1:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD/Board_Defs.h:238:0: warning: "digitalPinToAnalog" redefined [enabled by default]
 #define digitalPinToAnalog(P) ( ((P) > 14 ) ? digital_pin_to_analog_PGM[P] : digital_pin_to_analog_PGM[(14 - P) + 30] )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from sketch/sketch_dec16a.ino.cpp:1:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:244:0: note: this is the location of the previous definition
 #define digitalPinToAnalog(P) ( (((P) > 15) && ((P) < 32)) ? (P)-16 : NOT_ANALOG_PIN )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:263:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from sketch/sketch_dec16a.ino.cpp:1:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD/Board_Defs.h:244:0: warning: "analogInPinToChannel" redefined [enabled by default]
 #define analogInPinToChannel(P) ( analog_pin_to_channel_PGM[P]  )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from sketch/sketch_dec16a.ino.cpp:1:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:254:0: note: this is the location of the previous definition
 #define analogInPinToChannel(P) ( P )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:263:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WInterrupts.c:47:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD/Board_Defs.h:229:0: warning: "portRegisters" redefined [enabled by default]
 #define portRegisters(P) ( port_to_tris_PGM[P] )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WInterrupts.c:47:0:
[... 416 lines cut ...]
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:244:0: note: this is the location of the previous definition
 #define digitalPinToAnalog(P) ( (((P) > 15) && ((P) < 32)) ? (P)-16 : NOT_ANALOG_PIN )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:263:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/main.cpp:41:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/variants/Fubarino_SD/Board_Defs.h:244:0: warning: "analogInPinToChannel" redefined [enabled by default]
 #define analogInPinToChannel(P) ( analog_pin_to_channel_PGM[P]  )
 ^
In file included from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/WProgram.h:9:0,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/Arduino.h:4,
                 from /Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/main.cpp:41:
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/cores/pic32/pins_arduino.h:254:0: note: this is the location of the previous definition
 #define analogInPinToChannel(P) ( P )
 ^
cc1: fatal error: opening output file chipKIT-application-32MX795F512.ld.00: Permission denied
compilation terminated.
cc1: fatal error: opening output file chipKIT-application-32MX795F512.ld.01: Permission denied
compilation terminated.
/Users/rickanderson/Documents/Arduino/hardware/chipkit-core/pic32/compiler/pic32-tools/bin/bin/gcc/pic32mx/4.8.3/../../../../bin/pic32-ld: cannot open linker script file chipKIT-application-32MX795F512.ld.00: No such file or directory
collect2: error: ld returned 255 exit status
exit status 255
Error compiling.
```

majenkotech, 2015-12-16:
Most of that stuff is just because people forget to `#undef` things before `#define`ing them again.  I spent ages fixing that stuff in some of the boards I use commonly.  The only really relevant bits here are the permission denied messages where it's trying to write something to somewhere it shouldn't.

majenkotech, 2015-12-16:
This problem is specific to the Arduino IDE. It works perfectly in UECIDE. It must be to do with the `cwd` when the compiler binary is called.

EmbeddedMan, 2015-12-20:
Jason, what's the next step to tracking this one down? Is there anything we can do to assist? I think this is the biggest bocking issue at the moment-

jasonkajita, 2015-12-20:
I think that we will need to add `--no-cpp` to the linker command line for now. (`-Wl,--no-cpp` if passing to pic32-gcc for linking). This will disable the linker-script preprocessing feature until we figure out where we are trying to write the file.

ricklon, 2015-12-21:
I just tried with this command:
```
compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--no-cpp,-Map={build.path}/sketch.map,--gc-sections -mdebugger -mno-peripheral-libs -nostartfiles

```
I got a successful compile using it.

EmbeddedMan, 2015-12-21:
The workaround suggested by Jason appears to fix this problem for us, for now. Hopefully in future compiler versions we will be able to turn the linker-script preprocessing back on.

jasonkajita, 2016-01-07:
>The preprocessed scripts were placed indeed in the current working directory which on Mac was the root directory "/". This caused those permission denied errors.
>One solution to this would be the following change in platform.txt:
>`compiler.c.elf.flags={compiler.warning_flags} -Os -Wl,--save-gld={build.path}/sketch.ld,-Map={build.path}/sketch.map,--gc-sections -mdebugger -mno-peripheral-libs -nostartfiles`
>Passing the "--save-gld" option to the linker instructs it where to save the scripts.
>Marius Geanta

EmbeddedMan, 2016-01-26:
Need to confirm that this works on Mac properly.

EmbeddedMan, 2016-01-29:
I have confirmed that the fix Jason proposed (--save-gld=} is now implemented and works properly with no errors.

closed 2016-01-29 by EmbeddedMan