chipKIT32/chipKIT-core #118: Some chipKIT example sketchs missing

JacobChrist, 2015-12-17:
For example, blink that uses PIN_LED1, but other too...

Propose adding them to the chipkit-core/libraries directory.  These may mirror the Arduino examples when they are inferior to the chipKIT examples.

EmbeddedMan, 2015-12-17:
Yeah, I've thought of that. Don't you think it will confuse folks though?
And do we duplicate the entire Arduino example sketch tree, or just the
blink sketch?

ricklon, 2015-12-18:
If there is more time we should review all the examples. I think we can add
a few examples in each area showing a best practice for chipKit.

--Rick

EmbeddedMan, 2016-01-06:
Rick- I agree this is a good idea. But since we can't replace the default example sketches, how do we get people to use our chipKIT examples instead?

JacobChrist, 2016-02-07:
This is not just a Blink issue.  There were changes made to the SD library sketch to support several Digilent Boards as well as Arduino / Fubarino. Also, there was a simple parser example that Mike created that is sorely missing from the Arduino examples.

If directory hierarchies work I propose a directory called chipKIT-Built-In-Examples that mirrors the Arduino Built In Examples but only for examples that differ from the Arduino Examples.  I have a day and a half a week to help with this issue for the next two weeks.

ricklon, 2016-02-07:
I think we need to find a way to create a submenu. Just for the chipKIT
libraries. That would be easiest.

Also, if there are just general sketch improvements we can do some pull
requests.

Also, we might want to use a library prefix. I know it's not the best idea,
but some like "ck" can show the chipKIT libraries. Digilent does this with
DSPI. But it's company instead of platform oriented.

JacobChrist, 2016-02-07:
I'm trying to figure this out now... My class with the Fubarino SD starts in about 3 weeks and I would like to use some of the old examples.

ricklon, 2016-02-07:
I can dig back though and look for them.  I can help bring them back. They
might be able to be added in a special folder for sketches of certain kind.
Might be able to look at the example pattern and see we can add sub section
for chipKIT in the examples area.

--Rick

JacobChrist, 2016-02-07:
Okay, I got them working but its kinda ghetto.  I made a library folder and put in a chipKIT.h file (so that 
it gets picked up by the IDE).  Then I created an examples folder that has the examples.  Just unzip and add the following to the 

...\Arduino\hardware\chipkit-core\pic32\libraries

[chipKIT_Built_In.zip](https://github.com/chipKIT32/chipKIT-core/files/120880/chipKIT_Built_In.zip)

JacobChrist, 2016-02-07:
Its ghetto primarily, because now it looks like there is a library (in contributed) that shows up but doesn't really do anything or even exist.  Could cause confusion.

ricklon, 2016-02-07:
I think it's a good start. I'll look into this and see if there is a way to
make having sketches available less confusing.

JacobChrist, 2016-02-08:
I've added a leading and trailing underscore to my directory name for two reasons.

1. To make sure this is near the top of the examples list.
2. To make it stand out among "libraries"

So now its called: _chipKIT_Built_In_

Also, I was thinking that it would be good to add a comment in each example as to why it exists when there may be an equivalent Arduino example.  This should help to justify / guide us in to knowing which should stay and which should go.  It also would be a signifier to the reader as to why some chipKIT Built In examples exists but not all.