chipKIT32/chipKIT-core #149: pic32prog Fails to Upload when avrdude Succeeds

rei-vilo, 2016-01-08:
`pic32prog` fails to upload...

>
$ /Users/ReiVilo/Library/Arduino15/packages/chipKIT/tools/pic32prog/v1.0.01-pic32gcc/pic32prog -d /dev/cu.usbserial-A600CMUQ -e /private/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build76d3ba0842706beddde3a5ca178b86ee.tmp/chipKIT_blink.ino.hex
Programmer for Microchip PIC32 microcontrollers, Version 2.0.185
    Copyright: (C) 2011-2015 Serge Vakulenko
      Adapter: STK500v2 Bootloader
 Program area: 1d000000-1d1fffff
    Processor: Bootloader
 Flash memory: 2048 kbytes
  Boot memory: 80 kbytes
         Data: 6380 bytes
        Erase: Cannot erase chip.
>

...when `avrdude` succeeds.

>
$ /Applications/IDE/Mpide.app/Contents/Resources/Java/hardware/tools/avr/bin/avrdude -q -V -F -p32MX340F512H -C/Applications/IDE/Mpide.app/Contents/Resources/Java/hardware/tools/avr/etc/avrdude.conf -D -cstk500v2 -b115200 -P/dev/tty.usbserial-A600CMUQ -Uflash:w:/private/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build76d3ba0842706beddde3a5ca178b86ee.tmp/chipKIT_blink.ino.hex:i
>
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x504943
avrdude: reading input file "/private/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build76d3ba0842706beddde3a5ca178b86ee.tmp/chipKIT_blink.ino.hex"
avrdude: writing flash (9172 bytes):
avrdude: 9172 bytes of flash written
>
avrdude: safemode: Fuses OK
>
avrdude done.  Thank you.
>

Target is a **chipKIT uC32 board**. 

I'm using the [chipKIT Core](http://chipkit.net/wiki/index.php?title=ChipKIT_core#1.29_Auto_install_via_URL_from_within_Arduino_IDE) with the Arduino 1.6.7 IDE on Mac OS X 10.11.2.

rei-vilo, 2016-01-08:
The boot-loader of the chipKIT uC32 needs to be updated. 

See [PIC32-avrdude-bootloader](https://github.com/chipKIT32/PIC32-avrdude-bootloader).

closed 2016-01-08 by rei-vilo

aslankose, 2016-02-17:
I don't know what makes you believe this issue has ben resolved. I don't see anything regarding the solution. There is not a new boot-loader for chipKit uC32. If you have one please direct us to that. Because there is not any new file in the PIC32-avrdude-bootlader.

EmbeddedMan, 2016-02-17:
aslankose,

Digilent maintains their own bootloader code - we do not have any Digilent bootloaders in PIC32-avrdude-bootloader anymore. So you will have to go to their site, or ask them, and they will be happy to give you the latest uC32 bootloader, which should solve this problem for you.

aslankose, 2016-02-17:
Thank you EmbeddedMan