chipKIT32/chipKIT-core #153: DSDVOL not mounting SD card with WI-FIRE board

Nate42, 2016-01-10:
There appears to be an issue with DSDVOL (and by extension HTTPServer which depends on it) with the WI-FIRE board. I assume this must be related to the initialization of the SD card, because I have found that if I initialize the card using the old SD library, and then load new code using DSDVOL without power cycling, everything works fine. But if I then power cycle the board without changing anything, on restart DSDVOL can no longer mount the SD card. I've been poking around with the library code myself to see if I can find a fix, but no luck so far.

EmbeddedMan, 2016-01-10:
Keith, I'm going to point this one to you.

KeithV, 2016-01-10:
I don’t have this problem?

 

Does he have the latest HTTPServer? 

Does the DSDVOL example run?

 

KeithV

KeithV, 2016-01-10:
What is the error code that is returned from the mount? 

 

Does this happen with another brand of SD card?

KeithV, 2016-01-10:
I am unable to reproduce the problem.

 

I just took a brand new WiFIRE out of the box, formatted a scandisk card, put the IOServer files on the uSD card and with Arduino 1.6.7 and the 1.1.0 core successful programmed and ran the IO server on a WiFIRE. I then cycled power on the WiFIRE and started it again, all worked.

 

The only thing I can figure is there is that your FAT File system is corrupt on the uSD and the SD library doesn’t care and the DFATFS does. Try reformatting the uSD and put the IO server content back on.

 

KeithV

Nate42, 2016-01-10:
Hi all. Keith thanks for looking into this. 
I initially found this problem with HTTPServer, but SDWriteRead has the same behavior. The error I get is Failed to mount drive 0:! Error: 3
I am presently using Arduino IDE 1.6.7, and the just released 1.1 core, with a brand new 8GB Sandisk card, formatted to fat32. I have been there and done that with the multiple SD cards, reformatting, etc. Even tried using a Linux machine to remove Windows hidden indexing files. I would think if there was a problem with the FAT file system it would never work, rather than being able to "trick" it into working as I have been.  I've been fighting this for a while, and it was only recently that I realized 100% of the time I actually got it to work, I had run SD CardInfo as a sanity check first. Once I realized this I've reproduced it over and over again. Load CardInfo then load HTTPServer or SDWriteRead, works.  Power cycle, no longer works. Reload HTTPServer without first loading CardInfo, doesn't work. So I thought I was on to something there, but maybe not if you can't reproduce it.

Maybe I have a hardware issue? Although I'm not sure what it would be. I have a newish WIFire received about a month ago. When I get the chance I'll bust out the oscilloscope and see if I can get any insight as to if/how the SPI signaling behaves differently with the different methods.

Nate

EmbeddedMan, 2016-01-14:
Nate/Keith - what is the resolution path to closing this bug?

Nate42, 2016-01-14:
Given Keith's inability to replicate the bug, there's a good chance I may
have a hardware rather than software issue. I intend to take a closer look
with an oscilloscope but haven't had the time this week. If he's interested
in giving me contact info, perhaps this issue should be closed and Keith
and I should continue discussion outside of Github?

EmbeddedMan, 2016-01-15:
Nate - that sounds like a good plan. I'll close this issue, as it doesn't necessarily seem like it has to do with something wrong in chipKIT-core. If, at a later time, you find that the problem really is something that should be fixed in this code, please feel free to re-open the issue, or add a new one with new information. You and Keith can continue to try and troubleshoot your hardware.

KeithV, 2016-01-15:
Brian,

 

Can you give Nat all of the latest Arduino 1.6.7 and core 1.1.0 URLs so we can ensure he has the latest and greatest? A lot of stuff has been changing and I wonder if he got some half-finished thingie-magingie. 

 

Thanks

KeithV

EmbeddedMan, 2016-01-15:
Nate, and make sure you're using the very latest v1.1.0 code (this is what Keith is using I believe) : see here http://chipkit.net/wiki/index.php?title=ChipKIT_core

Nate42, 2016-01-19:
I think there was something to this after all. Some SD cards don't like to initialize properly if their SDO pin (SDI from the micro's perspective) isn't initially pulled high on startup. I was able to solve this problem for my particular card by enabling the internal pullup with CNPUBbits.CNPUB10 = 1; in the initial setup.  I tried this with two different SD cards that didn't work previously and now they seem to work fine.

EmbeddedMan, 2016-01-19:
Nate, this is a great suggestion.

Keith, do you concur that adding  this internal pullup in the setup is a good idea? If so, would you like to submit a PR with this change so we can get it into core?

KeithV, 2016-01-19:
Brian / Nat,

 

I am not sure. Great work detecting the issue… here is the problem. SPI would not want MISO pin to have a pullup put on by the receiver; maybe the SDO owner yes, but probably not the SDI owner. Now, I have to look at the SD card spec as that may have additional requirements as other things may be happening. It is interesting that the Bit Banged (SD) version seems to work, but the hardware version does not. This to me says that the specific brand of SD card may not be following the spec.

 

But, there is another way to look at this… who cares what any of the specs say if this is what it takes to make it reliable. The damage to turning on the pullup resistor is that it will slow down the SPI signal, maybe too slow. However, I think we are running them currently slow enough that it will work. Clearly that is what Nat is indicating.

 

Brian; what is your opinion?

 

KeithV

KeithV, 2016-01-19:
Nat, can you tell me the brands of the SD cards you are having trouble with?

Nate42, 2016-01-19:
One was a brand new Sandisk 8 GB, another is a Lexar 1GB that is probably 5
years old or so. I wouldn't expect it to impact speed since the SD card is
ultimately still actively driving it high, its not relying on the pullup
like an open drain situation. The only downside I can think of is very
slight increase in current consumption since the card has to pull down the
resistor. Its a weak pullup so it doesn't have much impact. Waveforms still
looked clean on the scope.

KeithV, 2016-01-19:
I looked at the spec and there is one possibility... when we initialize is CS high for 80 SPI clock cycles before the init command? This would create a problem if CS is low for this timeframe.
BTW: I use scandisk SD cards no problem.

Nate42, 2016-01-19:
CS was doing what its supposed to I did verify that.

EmbeddedMan, 2016-01-19:
Is it the kind of thing that we can turn on just when we are initializing,
and then turn off once we're "talking" to the card? Would that help reduce
the potential impact at all?

I'm all for doing what we know works - I do know that in a previous job,
where we had to create an embedded module that could properly boot from
_any_ SD card, we ended up literally buying one of everything, and testing
them all (we had several hundred in a big notebook), and anytime one of
them didn't work for some reason (which was more frequent than you'd
expect) we'd have to change some of our code to match what made the card
work. So I'm certainly not against doing what we know will work, even if it
goes against the spec a bit.

*Brian

KeithV, 2016-01-19:
Before I just go off and do something, I would like to know why? A few things, what motivated you to turn on the pullup? And can you debug? I would like to debug DSDVOL.cpp in disk_initialize and know what is failing. Is if faling the card detect? is it failing setting the block size, is it failing the clock speed change? Is it failing voltage range? Is it timing out?

KeithV, 2016-01-19:
I would be great if I could repro it here because I would understand it better and then I could make an informed decision. My fear is there is a more intrinsic problem. One big difference between SD and FATFS/DSVOL is I change the SPI clock speed. You must start out slow and then you can bump the speed up once you initialize. The SD implementation just ran a a constant too fast speed for init; but seemed to work. I attempted to follow the spec, but maybe that is causing a problem? If so, I need to fix that.

KeithV, 2016-01-19:
But yes I could add code to turn on the pullup just for the init; of course we would have to fix the variants for the other boards as right now only the Uno32, uC32, and WiFIRE handing the pullup code correctly. We hit that problem that the changeNotice tables are not correct on most MX boards.

Nate42, 2016-01-19:
I have an ICD3 at work that I suppose I could debug with, I haven't gotten
to the point where I've tried to actually do that though. In this instance
though I can tell you where and how it is failing. If you probe the sd
cards SDO line, there is no signal at all the sd does not respond to the
initial reset command. Exact same behavior as if no card was present. So it
is presumably timing out due to no response.

Nate42, 2016-01-19:
Oh, and since its failing to respond to the very first command, this I
believe takes place before you ramp up the speed.

Nate42, 2016-01-19:
My inspiration to try the pullup came from this random internet link: http://elm-chan.org/docs/mmc/mmc_e.html which suggested a pullup on the SD's DO line. I figured it couldn't hurt, and was frankly pretty surprised when it seemed to work. I tried with and without the pullup and results were repeatable. 

I'm an EE by trade, so I somewhat know what I'm doing, but interfacing with an SD card is new for me.

KeithV, 2016-01-19:
What you should see is 80 clock cycles at an SPI speed of (I think) 400KHz, 1s should be clocked to the SD card with CS high, Then you should see another 8 clock, with CS High, Then CS low with 8 more 1s clocked out. And here is were we probably have the problem... You should see 8 more 1s clocked out, but we expect 1s (SDI to be high) clocked back in. This could be the problem because we have not put the SD in idle state yet... although it should be clocking ones back. If the SDO on the SD card is still in try state, this would be board dependent as it is just floating. Can you see if MISO is low during the time CS is low, or at least the 2 8 1s that get shifted out while CS is low?

KeithV, 2016-01-19:
If CS goes low on the SD, SDI on the SD should go high; not stay try stated..... In fact, the SD card has 300ms to indicate it is not busy by setting SDI high, we wait 500ms... The card is indicating it is still busy. Funny the old SD implementation is less forgiving than the new one. Here is what I will try to do. I will set the pullup low so any tri-stating will be exposed and I will get a new scan disk card and give this a whirl. I want to understand this.

KeithV, 2016-01-19:
OH, by no means do I doubt your abilities. I suspect it is just one of those things where various people did various things; now we have to deal with reality. My guess is, the SD card is sloppy before it is put in the idle state. A bug in the SD hardware, but we still have to work.

KeithV, 2016-01-19:
I read your research and they very clearly say a pullup is needed. I suspect this is experience talking. So I suspect we need to do that. This is a hardware issue I did not know about. This is a good call.

Nate42, 2016-01-19:
No worries I didn't suspect you doubted my abilities. Rather just letting you know of my capabilities since I'm happy to help with stuff like this. And as I said I really didn't know anything about SD cards prior to this exercise, but I know a teensy bit now.  If anything comes up in the future you'd like me to take a look at let me know.

KeithV, 2016-01-19:
Can you send me an email KeithV@Digilentinc.com; I have some files for you to try.

Nate42, 2016-01-25:
The files Keith sent me seem to be working without issue. I haven't tried the build from his pull request yet but I assume its the same? If so I'm fine with closing this, for good this time.

EmbeddedMan, 2016-01-25:
Nate,

Yup, they should be identical. Thanks so much for testing it out!

*Brian

EmbeddedMan, 2016-01-26:
Keith's PR with the changes discussed in this issue has now been merged into master.

closed 2016-01-26 by EmbeddedMan

alniju, 2018-12-23:
There is the same issue with version 2.0.6 of core and version C of WiFire, had to change version of core to 1.4.3. now it works

majenkotech, 2018-12-23:
This is because of a change made to generalise the SDVOL library.  Initially it had hard-coded (and severely broken) code in it to enable the internal pullup on the MSIO line of the SD card, bacuse the WiFire design lacks a proper pullup on that pin.  The SD constructor was changed from taking a broken flag (incorrectly named a pullup pin and with strange logic behind it) to instead take a chip-select pin number.  This allowed the library to be used on other boards since there was no provision to provide a chip select pin, which was just plain stupid.  It was so stupid, in fact, that all the board definitions passed the CS pin number as that parameter already, even though that parameter wasn't the CS pin number, just a pullup flag. Setting the chip select was actually being done elsewhere in a macro in the board definitions, which most board definitions lacked. So the decision was made to make the library act how people expected it to act, and how the board definitions were expecting it to act, and anything strange that is specific to only a couple of situations should be left up to the user to arrange in their sketch.  

So now you have to *manually* set the internal pullup to get around the lack of a proper pullup in the WiFire design.  Adding:

    pinMode(53, INPUT_PULLUP);

just before you try and initialize the card should fix it.

Note: this is not needed for all SD cards. Some need it, some don't. It depends on the card.