chipKIT32/chipKIT-core #155: Proposal: process platform.txt before deployment

ricklon, 2016-01-10:
Currently, we're keeping track of 2 platform.txt files. Likely 3 because of MPIDE. But the case of chipKIT-core the two we're tracking are pretty similar. It's possible to use platform.txt as a template and substitute the correct settings when built for URL or Manual install. The difference between files is minimal:
```
23c23
< compiler.path={runtime.hardware.path}/pic32/compiler/pic32-tools/bin/
---
> compiler.path={runtime.tools.pic32-tools.path}/bin/
117c117
< tools.pic32prog.path={runtime.hardware.path}/pic32/tools/bin
---
> tools.pic32prog.path={runtime.tools.pic32prog.path}
```
We would pass a variable about whether build is for URL or manual install. Then correct values would be selected. The search and replace would occur putting the correct values in platform.txt.

If the regex replace task can work this would be possible:
https://ant.apache.org/manual/Tasks/replaceregexp.html

ricklon, 2016-01-10:
Ok. I just did a test and it's possible to do this!  I'm not sure when these replace tools got added to ANT but yay, it works.

```
<target name="test" >
	  <replaceregexp file="pic32/platform.txt"
		  match="runtime"
		  replace="oldtime"
		  byline="true"
		  />
  </target>
```

EmbeddedMan, 2016-01-10:
That's great Rick! I'm just uploading a complete set of manual builds right
now that will be the v1.1.0 manual builds. I'll link to them in the Wiki
later this afternoon.

*Brian

EmbeddedMan, 2016-01-13:
Rick, do you have a PR coming that implements this change to build.xml? I'd love to give it a test.

ricklon, 2016-01-13:
I can do it but it needs to be Thursday or Friday. Just finished a 2 day
conference presenting, and running makerspace booth. I need a day or so to
catch up.

--Rick

EmbeddedMan, 2016-01-13:
No worries, and no rush. If I get a chance I might peck at it, and I'll
update this issue if I do. Otherwise it's all yours, whenever you have some
time.

*Brian

rianders, 2016-01-25:
I just created the initial code. But need to figure out the process. We want to keep one mater file. Should it be the manual install or platformURL. The source file would be copied and the regex would be applied to the copy.

Example regex replacement:
```
<target name="updatePlatform" >
      <replaceregexp file="pic32/platformURL.txt"
          match="\{runtime.tools.pic32-tools.path\}\/bin\/"
          replace="\{runtime.hardware.path\}\/pic32\/compiler\/pic32-tools\/bin\/"
          byline="true"
      />
      <replaceregexp file="pic32/platformURL.txt"
          match="\{runtime.tools.pic32prog.path\}"
          replace="\{runtime.hardware.path\}\/pic32\/tools\/bin"
          byline="true"
      />
  </target>
```

EmbeddedMan, 2016-01-25:
Rick,

Sweet. I would suggest that the manual install be the 'normal' one, and
that the regex replacement create the URL install version.

*Brian

ricklon, 2016-01-26:
Can do!

EmbeddedMan, 2016-02-15:
Rick, any update on this task? Are we getting close?

rianders, 2016-02-15:
I figured it out.  I'll get the update in today.

We use the stand alone version as the main version. Then deploy it into the
dist directly and then run the replacement per OS directory.

--Rick

EmbeddedMan, 2016-02-21:
Rick, any update on this?

ricklon, 2016-02-22:
I've got a fix for this ready. We don't have a specific build target for the web distro. So I added a new target for that. It does the same thing as the "ant dist" there is now "ant dist-url"  The process replaces the necessary lines in platform.txt.

EmbeddedMan, 2016-02-22:
Great! Have you sent in a PR for it so I can test it out?

ricklon, 2016-02-22:
Oops. I meant "ant dist-all-url" as well as the individual commands like "ant dist-macosx-url" etc.

I'm running one last test. If there are no errors I'll get the pull request done.

ricklon, 2016-02-22:
I got the pull request in. Some review is needed to make sure I solved the correct problem. ;-P

ricklon, 2016-02-22:
Arg. My solution solves the problem for each platform as if each one was different. But we were creating the web/url distro by downloading the project and then uploading the result into releases. 

So this means we should keep the project default platform.txt to the url/web version. Then change it for the standalone. If we had access to the github api from ant we could do this the by leaving it as the manual version.

I just need to reverse the search a replace.

EmbeddedMan, 2016-02-25:
I've done both types of build with the code change you made, Rick, and the output looks correct to me. I need to test it via a real URL install yet to make sure that there are no problem. But so far it looks great!

EmbeddedMan, 2016-02-26:
I'm going to close this issue because I the output looks correct to me, and the original task is now complete. Thanks Rick!

closed 2016-02-26 by EmbeddedMan