chipKIT32/chipKIT-core #165: Put all libraries into archives

EmbeddedMan, 2016-02-01:
When compiling, the compiler will have a much easier time removing unused code if we compile all of our libraries (i.e. everything except the sketch) to an archive first, and then pass them all in to the linker.

So this task will involve working out how to modify the recipes in platform.txt  to achieve this goal.

xxxajk, 2016-02-14:
I had to muck with the recipes for this and some other stuff. 
The compile recipes are broken they are missing some defines when compiling the sketch, making inline code compile or adjusting a sketch to be universal impossible,
I did it before with an older version, but they never took ANY of the pull requests.
So I just nuked it.
I'll re-do it and you are welcome to test it.

EmbeddedMan, 2016-02-15:
Andrew, that would be awesome. Thanks for any help-