chipKIT32/chipKIT-core #169: Issue with the wifire examples

Metavix, 2016-02-12:
When you comment "//#define USE_WPA2_PASSPHRASE" and uncomment another type of connection arduino show a errror like this:

WiFiTCPEchoServer:110: error: 'key' does not name a type
      key = { 0x27, 0x2C, 0x89, 0xCC, 0xE9, 0x56, 0x31, 0x1E,
      ^
/tmp/arduino_cb69d79a5eabb78daa50e5450124e7ff/WiFiTCPEchoServer.ino: In function 'void loop()':
WiFiTCPEchoServer:114: error: 'key' was not declared in this scope
     #define WiFiConnectMacro() deIPcK.wfConnect(szSsid, key, &status)
                                                         ^
/tmp/arduino_cb69d79a5eabb78daa50e5450124e7ff/WiFiTCPEchoServer.ino:236:16: note: in expansion of macro 'WiFiConnectMacro'
             if(WiFiConnectMacro())
                ^
exit status 255
'key' does not name a type

KeithV, 2016-02-18:
Just remove the "DEWFcK::". When we moved from DWIFIcK to DEWFcK the fundamental types come from the underlying IP includes and are no longer part of the WiFi class. 

These examples need fixed... thanks.

Metavix, 2016-02-18:
No problem you'r welcome