chipKIT32/chipKIT-core #179: Lenny generates warning on any board compile

EmbeddedMan, 2016-02-23:
Any time you compile any chipKIT sketch for any chipKIT board, you get the following warning in the Arduino IDE: Board chipkit-core:pic32:lenny doesn't define a 'build.board' preference. Auto-set to: PIC32_LENNY

majenkotech, 2016-02-23:
If there's anything wrong with it now, blame the Fubarino Mini. I copied that config and lennyfied it.

EmbeddedMan, 2016-02-23:
Ha! There's probably a ton wrong with the Mini. Thanks for the PR -

EmbeddedMan, 2016-02-24:
Matt - I just tested out the new Lenny support, and it appears to work
great. I love this little board.

If you turn on all warnings, you'll see several in the Lenny variant files
related to static not being at the beginning of a definition, and an unused
parameter or two. Nothing major though.

EmbeddedMan, 2016-02-24:
Confirmed fixed!

closed 2016-02-24 by EmbeddedMan

majenkotech, 2016-02-24:
They must be specific to the Arduino IDE. I don't get those warnings in UECIDE with -Wall.  Do they turn on any special specific warning flags?

EmbeddedMan, 2016-02-24:
Yes - in platform.txt we now have
compiler.warning_flags.all=-Wall -Wextra

So when a user selects "All" for the warning level, those are the options
that get used.