chipKIT32/chipKIT-core #189: Fubarino Mini Minimal Sketch using DSPI fails to compile

ricklon, 2016-02-26:
```
#include <DSPI.h>

DSPI0 spi;

void setup() {
  spi.begin();
  spi.setSpeed(250000);
  spi.enableInterruptTransfer();
}
void loop() {
}
```

error:
```

/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp:1149:1: error: 'DSPI1' does not name a type
 DSPI1::DSPI1() : DSPI(_DSPI1_MISO_PIN, _DSPI1_MOSI_PIN, _DSPI1_MISO_IN, _DSPI1_MOSI_OUT)

 ^
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp:1181:1: error: 'DSPI1' has not been declared
 DSPI1::enableInterruptTransfer() {

 ^
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp: In function 'void enableInterruptTransfer()':
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp:1183:11: error: invalid use of 'this' in non-member function
  pdspi1 = this;

           ^
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp:1184:32: error: cannot call member function 'void DSPI::enableInterruptTransfer()' without object
  DSPI::enableInterruptTransfer();

                                ^
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp: At global scope:
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp:1204:1: error: 'DSPI1' has not been declared
 DSPI1::disableInterruptTransfer() {

 ^
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp: In function 'void disableInterruptTransfer()':
/Users/rickanderson/Library/Arduino15/packages/chipKIT/hardware/pic32/1.1.0/libraries/DSPI/DSPI.cpp:1207:33: error: cannot call member function 'void DSPI::disableInterruptTransfer()' without object
  DSPI::disableInterruptTransfer();

                                 ^
exit status 255
Error compiling.

```

EmbeddedMan, 2016-02-26:
Well, that stinks. I'll work on fixing this.

EmbeddedMan, 2016-02-26:
Rick, I get zero errors when I use the latest chipKIT-core, with the sketch above, compiled for a FBMini.

EmbeddedMan, 2016-02-26:
When I compile with release 1.1.0, then I get the errors you are getting. There has been one commit from Keith that fixed some things that is probably the reason master works when 1.1.0 doesn't. 

So I believe this problem is solved with the very latest code in GitHub. Can you test with the latest to see if it fixes for you too?

ricklon, 2016-02-26:
I will test that. :-)

ricklon, 2016-02-26:
This works in current master.

EmbeddedMan, 2016-02-26:
OK, good. I'll close this now, and we can do a new release whenever we feel there's a strong reason to do so (maybe getting this fix out to folks is a good reason?)

closed 2016-02-26 by EmbeddedMan