chipKIT32/chipKIT-core #190: I2C write lock up

tsupplis, 2016-02-28:
The following program locks up in the Wire:write code (Tested on DP32, UC32, chipKIT-core 1.1.0).

```
#include <Wire.h>

void setup()   {    
  Serial.begin(9600);
  Serial.println("Wire.begin");
  Wire.begin();
  delay(1000);
}

void loop() {
  Serial.println("Wire.beginTransmission");
  Wire.beginTransmission(0x3C);
  Serial.println("Wire.write");
  Wire.write(0x00);
  Serial.println("Wire.write");
  Wire.write(0xAE);
  Serial.println("Wire.endTransmission");
  Wire.endTransmission();
  delay(1000);
}
```
The lock up happens around line 314 in DTWI.cpp:
`    pregIec->set = bitMASK; // turn interrutps back on
`

KeithV, 2016-03-09:
Do you have any I2C device plugged into the board? Or are you just open loop expecting it to fail? This is the problem with the Wire class, it doesn't return error.... so you just keep going if nothing is out there.

tsupplis, 2016-03-09:
Both, with or without a device, it blocks. I am using an I2C OLED screen. It works well with an alternative library that uses the i2c registers directly.

tsupplis, 2016-03-09:
The library I used as a test point is OLED_I2C on
http://www.rinkydinkelectronics.com/library.php?id=80

KeithV, 2016-03-09:
I just tried it on an uC32 and it runs..... nothing attached. Your code exactly.

Clearly not working as nothing is attached, But that is because the Wire library API are pretty brain dead.

KeithV

Wire.begin
Wire.beginTransmission
Wire.write
Wire.write
Wire.endTransmission
Wire.beginTransmission
Wire.write
Wire.write
Wire.endTransmission
Wire.beginTransmission
Wire.write
Wire.write
Wire.endTransmission
Wire.beginTransmission
Wire.write
Wire.write
Wire.endTransmission
Wire.beginTransmission
Wire.write
Wire.write
Wire.endTransmission
Wire.beginTransmission
Wire.write
Wire.write
Wire.endTransmission
Wire.beginTransmission
Wire.write
Wire.write
Wire.endTransmission
Wire.beginTransmission

KeithV, 2016-03-09:
Plugged in a BasicIO sheild, it continues to run; not that the OLED comes on, but it doesn't hang.

tsupplis, 2016-03-09:
Ok, let me try again with the device attached. Would you recommend bypassing Wire? I am happy to code it from scratch using plib. I am happy to use the registers directly.

tsupplis, 2016-03-09:
Interestingly you were able to run the program wihout attached and everything goes fine (if we ignore the errors). whether in my case it locks almost strait away on the line above...

KeithV, 2016-03-09:
Not sure how to help if I can't repro. Can you ask Brian to give it a go? He had me make some changes for compatibility so he is up to speed on this. Maybe if we can get a second point of failure we can debug.
I WOULD RECOMMEND USING DTWI DIRECTLY!! There is a comprehensive example in the examples directory. You will be able to .... check for errors ..... :-)

tsupplis, 2016-03-09:
Thank's for your advice Keith. I suppose close is in order.

closed 2016-03-09 by tsupplis