chipKIT32/chipKIT-core #191: Distribution includes twice gcc and binutils tools

tsupplis, 2016-02-29:
I just noticed that bin/bin, bin/device_files and bin/bin/gcc directory include duplicates under chipKIT/tools/pic32-tools/4.8.3-pic32gcc
- bin/device_files has the same content as pic32mx/device_files
- bin/bin seems to overlap but not full with bin

tsupplis, 2016-02-29:
Or is it mirroring the xc32 distribution?

EmbeddedMan, 2016-02-29:
Yes, it mirrors how XC32 is set up. I still don't know why we (or XC32)
needs two copies of all that though. I'll look into it.

EmbeddedMan, 2016-02-29:
Jason - is there a reason that there are multiple copies of some of these files in the chipKIT compiler?

tsupplis, 2016-02-29:
Anything I can do to help?

EmbeddedMan, 2016-02-29:
Not at this point - I've assigned this issue to Jason Kajita, who knows a
lot more about the compiler than I do and can give us some background on
why it is this way, and maybe if it can be changed to be 'slimmed down' at
all. Thanks for bringing this up though - I've thought the same thing in
the past.

jasonkajita, 2016-04-28:
The standard GCC build process is creating them. We will see if we can remove them with a post-build script or something like that.

EmbeddedMan, 2016-04-28:
Jason- that would be awesome!

jasonkajita, 2016-04-29:
We can get rid of the copy of the device_files in the pic32mx/bin directory, but we need to keep them in both bin and pic32mx directories for now. @mariusgeanta

EmbeddedMan, 2016-04-29:
What about the bin/bin vs bin overlap? Those are some larger files and
would be really nice not to have duplicates-