chipKIT32/chipKIT-core #200: All build tools are missing

VisualMicro, 2016-04-03:
The pic32\compiler folder is missing

EmbeddedMan, 2016-04-03:
If you downloaded chipKIT-core via the Arduino IDE (i.e. through the Boards Manager) then the compiler and tools (like pic32prog) have to be installed in a different location. 

They will be in 
C:\Users\Brian\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc
(top level path will change based on your OS)

The chipKIT-core files will be here:
C:\Users\Brian\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.1.0

Does that solve the problem for you?

VisualMicro, 2016-04-03:
The problem was that I was confused by the instructions at the root of this site. 

I have since found the wiki page that describes the updated integration to the Arduino Ide board manager. I have now used the index.json and installed the chipKIT core which correctly includes the compiler as described in the platform.txt.

http://chipkit.net/wiki/index.php?title=ChipKIT_core

So everything just worked out of the box when I used the Board Manager to install. All chipKIT boards are compiling fine in both the Arduino Ide and Visual Studio.

Great job thanks!

closed 2016-04-03 by VisualMicro

EmbeddedMan, 2016-04-03:
You raise a good point - I'm going to update the readme.md file to point to the new Wiki documentation too. Thanks!