chipKIT32/chipKIT-core #201: Minor documentation change

VisualMicro, 2016-04-03:
In the install instructions for Arduino 1.6.x naming the {sketchFolder} is misleading because this suggests the current program/sketch/project folder.

copy the dist/{OS} zip file to {SketchFolder}/Arduino/hardware

The SketchBook folder is where all sketch projects are normally stored and that is where the user hardware folder should be located/created.

copy the dist/{OS} zip file to {SketchBookFolder}/Arduino/hardware

Thanks

VisualMicro, 2016-04-03:
After reading some more these instructions seem to have been superceeded by integration with the Arduino 1.6.7+ Board Manager. 

So maybe these instructions can be updated to

**Arduino 1.6.7+**
Add this board support url to the Arduino Ide File>Preferences. 

`https://raw.githubusercontent.com/chipKIT32/chipKIT-core/master/package_chipkit_index.json`

After adding the url you will find the ChipKIT boards can be installed using the "Boards>Board Manager" in the Ide.

EmbeddedMan, 2016-04-03:
Which instructions are you referring to? If you can give me a pointer to it, I'll get it updated.

Thanks for pointing it out!

VisualMicro, 2016-04-03:
The root of this repository shows readme.md 

https://github.com/chipKIT32/chipKIT-core

Thanks v much

closed 2016-04-03 by VisualMicro