chipKIT32/chipKIT-core #204: WiFire does not run with v1.1.0 of core

EmbeddedMan, 2016-04-07:
If you try the following sketch with chipKIT-core v1.1.0 (URL install) and a WiFire board, the board will never leave bootloader mode after uploading. It might be that the bootloader doesn't think a valid program is loaded, as if you reset the board, it will immediately go into bootloader mode and stay there.

#include <DSPI.h>
 
void setup() {
  // put your setup code here, to run once:
  pinMode(PIN_LED1, OUTPUT);
  
}
 
void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(PIN_LED1, HIGH);
  delay(5000);
  digitalWrite(PIN_LED1, LOW);
  delay(5000);
}

KeithV, 2016-04-15:
This is a bug in Arduino IDE 1.6.8. 1.6.8 continuously pulses DTR thus continuously resets the MCU. If you shut down the IDE the freaky blinking goes away and the sketch (whatever is loaded) will run normally. 

Go back to Arduino IDE 1.6.7 and complain to the Arduino folks for breaking ALL FTDI boards. Of course the Arduino Uno uses their own USB/Serial implementation with an ATMega16U2.

EmbeddedMan, 2016-04-18:
I have not yet found the issue in Arduino's GH issue list yet, but this has been fixed already in their IDE. If you go grab a nightly build, you will not have the problem - the latest nightly is able to build and upload and run on WiFire just fine, multiple times in a row. The reset is not being toggled. So we just have to wait for 1.6.9, hopefully that will get released soon, and tell chipKIT users NOT to use 1.6.8 at all.

EmbeddedMan, 2016-04-30:
Another easy workaround to this problem - after you are done uploading your new sketch to your board, simply open the serial monitor in the Arduino 1.6.8 IDE - that prevents the IDE from constantly re-enumerating the bus and resetting the FTDI chips.

EmbeddedMan, 2016-05-10:
The latest version of Arduino IDE is out - v1.6.9, and it has fixed this issue, and it is a listed changed in the IDE. For us, this is done now  - we just need to make sure to steer users away from 1.6.8.

closed 2016-05-10 by EmbeddedMan