chipKIT32/chipKIT-core #208: Can't Upload to uC32 w/o board reset

mr337, 2016-04-18:
I'm having issues uploading the a sketch to the uC32.
```
Programmer for Microchip PIC32 microcontrollers, Version 2.0.186
    Copyright: (C) 2011-2015 Serge Vakulenko

No target found.
```

If I hit build and quickly hit the reset on the uC32
```
Programmer for Microchip PIC32 microcontrollers, Version 2.0.186
    Copyright: (C) 2011-2015 Serge Vakulenko
      Adapter: STK500v2 Bootloader
 Program area: 1d000000-1d1fffff
    Processor: Bootloader
 Flash memory: 2048 kbytes
  Boot memory: 80 kbytes
         Data: 6204 bytes
        Erase: done
Program flash: ........######### done
 Verify flash: ........######## done
 Program rate: 3017 bytes per second
```

Here is what I have done to get things working to the current state.
* Upload the community firmware from https://github.com/chipKIT32/PIC32-avrdude-bootloader from my SO Mac. Wouldn't work on my Linux laptop , got `Could not erase board` message no matter what I did.
* Grabbed the latest release of https://github.com/sergev/pic32prog, `v2.0.196` iirc - to no effect
* The now discontinued MPIE does upload properly without needing to be rest, from my Linux laptop, so assuming pressing reset isn't normal part of build process :laughing: 
* Uninstalled stable core and installed latest from Jenkins build.

By coincidence I realized that when I build and reset together I can get it working. I thought maybe it was something relating to grabbing the device, but I think the reset button only resets the processor, not the FTDI chip on there. Dmesg shows no new devices leaving or getting added when reset button it pressed. Pretty confident nothing is grabbing that device.

Setup:
Arch Linux x86_64
Arduino 1.67
Core from Jenkins latest build (chipkit-core-linux64.16778039.HEAD-v1.1.0-56-gd625813.zip	104.43 MB)

EmbeddedMan, 2016-04-20:
I wonder if this issue is related to issue #204 which has to do with the Arduino IDE improperly resetting FTDI based boards. We've found that this issue is fixed in the latest Arduino IDE nightlies. Could you try using a recent nightly IDE build and see if that has any effect on your ability to upload?

Also keep in mind that the Digilent boards are not supported any more by the community bootloader projects - you need to get the bootloader HEX files (and source if you ask nicely) directly from Digilent for their boards.

You are correct in that the reset button only resets the processor, not the FTDI chip. Pressing the reset button on the uC32 will reset the CPU, jump to the bootloader, and wait for approximately 2 seconds for 'bootloader' data from the PC over FTDI before trying to jump to any programmed sketch in flash.

The interesting thing to me here is that MPIDE works for you, but Arduino IDE 1.6.7 does not. I know a number of other folks using IDE 1.6.7 with uC32 and it's working for them, so I wonder if there isn't some other interaction going on here. Maybe it's because you're on a 64 bit version of Linux? Although I don't know enough about Linux to know why that would be a problem. It's just most of us are Windows and Mac folks, so those platforms get more time and things get found and fixed faster on them typically.

majenkotech, 2016-04-20:
FTDI and Arduino have always been a bit of a chore. MPIDE and UECIDE have specific reset code embedded in them to toggle the DTR properly. I am not sure what the Arduino IDE has these days, but it sounds like they have been having real fun with it on there.

I have found that sometimes the /dev/ttyUSBx port "forgets" its DTR handling occasionally. This is usually due to some external software turning it off.  It's possible to turn it back on again with

    $ stty -F /dev/ttyUSB0 hupcl

at which point reset starts working again. 

If you have some form of oscilloscope or logic analyser it could be useful to get a trace of the RESET signal while it's failing - see if there is any activity at all on there, though it sounds to me like there won't be.

mr337, 2016-04-21:
@EmbeddedMan I tried the Arduino nightly, I guess hourly now and with no success. Still same issue and same work around. All while MPIE still worked.

I would like to know more about the bootloader as I'm a bit uneasy using this for my current or future projects. From what I have gathered it looks like a gentleman named Kevin has been releasing community bootloaders that can be uploaded via a sketch. If this has now changed, is Diligent releasing bootloaders that we can just download? Are we at the mercy that they might revoke access the to bootloaders rendering the board incomplete? Does this also require getting a programmer or can be programmed via a sketch?

Also I appreciate answering back even though I'm on a Linux system. I have no interest to move away from it any time soon, so working with Linux is a must for me!

@majenkotech Your command did the trick. After I set that it has reliably been able to upload the blink sketch with Ardino 1.67 and nightly. It looks like it has to be set once and so far has been persistent between connecting and disconnecting.

I'm just now entering the microcontrollers for personal projects and on my to buy list is a logic analyzer and oscilloscope. All the good ones are a pretty penny!

majenkotech, 2016-04-21:
I'm 100% Linux through and through, so support for Linux will be around as long as I am around ;)

There is a community variant of the bootloader which we all use on our own boards - it's only Digilent that are tight with them.  I have a fork of the bootloader repo that has all the DIgilent configurations in it that you can download: https://github.com/majenkotech/PIC32-avrdude-bootloader/tree/digilent/BootloadersCurrent-hex

I am also in the process of building a completely generic bootloader update sketch to allow updating without a hardware programmer - although a hardware programmer is certainly always a handy thing to have to hand.  Work in progress is here: https://github.com/majenkotech/bootloaderUpdater

It's not to be considered release quality yet - you should only use it if you have a hardware programmer to hand to "undo" whatever it breaks ;)

As for logic analysers, they don't have to cost a bomb.  I use one of these: http://eud.dx.com/product/logic-analyzer-w-dupont-lines-and-usb-cable-for-scm-black-844148945#.VxkzU2ErKCw - works beautifully with Sigrok and PulseView on Linux.

mr337, 2016-04-25:
That is awesome news. I was a bit worried about committing to going with the uC32. The whole bootloader game doesn't leave a good taste.

I'll be getting a hardware programmer along with a jtag so I have proper tools. If you need some testing let me know!

That digital analyzer is dead cheap from what I was looking, and bonus to work on Linux. Making a purchase now. You don't happen to know about a $10 oscilloscope :D :D

I'm closing this as it isn't related to chipkit but to a tty setting.

closed 2016-04-25 by mr337