chipKIT32/chipKIT-core #213: Non-PPS PIC32s fail to compiled SoftwareSerial.h from chipKIT-core 1.1.0

EmbeddedMan, 2016-05-10:
The CN_table[] array fails to compile properly - it looks like the _IOPORT_PC, _IOPORT_PB, etc. definitions are no longer available. This should be fixed and tested on all chipKIT boards to make sure it compiles and an example software serial sketch runs properly.

JacobChrist, 2016-07-12:
Does anyone know why the _IOPORT_PC, _IOPORT_PB definitions were removed (if they were removed)?  If they were not removed then it just might be that the class can't see them.
If they were removed maybe we can patch this for now just to get this class working on all board again.

hormigaAzul, 2016-07-13:
Possible patch https://github.com/chipKIT32/chipKIT-core/pull/236

JacobChrist, 2016-07-14:
I may have time tomorrow to test on a Fubarino SD and a Quick240 (Max32
Equivalent).

Jacob

EmbeddedMan, 2016-07-14:
Jacob if you can test this out and make sure it it doesn't break anything, I'll get it merged.

kosme, 2016-07-14:
I could test it on ChipKIT Uno32

kosme, 2016-07-21:
Worked nicely for me.

JacobChrist, 2016-07-21:
I haven't gotten around to testing this yet.  As Brian noted somewhere else
the fix seems pretty straight forward.  I don't know if this is the right
place to put this?

One thing I can say is that if it even compiles then its working 100%
better than before.

Jacob

EmbeddedMan, 2016-07-25:
I'm going to close this issue, as the PR has been merged and the code appears to work. If anybody finds a problem with the change, please re-open this issue.

closed 2016-07-25 by EmbeddedMan