chipKIT32/chipKIT-core #218: No DSPI1 defined in Fubarino SD

nanoq, 2016-05-17:
There is only 1 Port defined with 'NUM_DSPI_PORTS 1' in variants/Fubarino SD/Boards_Defs.h since ChipKit-Core 1.1.0. (In MPIDE 0023 and MPIDE 0150 are 2 DSPI Ports!) I have the need to use two SPI Ports in my project. What can I do?
Kind regardes - nanoq

EmbeddedMan, 2016-05-17:
You are correct. The reason for this is that the 'base' level of Fubarino SD (v1.4) produced by SeeedStudios (which has the PIC32MX440) only has 1 SPI port. That processor doesn't have more than one SPI port on it. The reason that the MPIDE versions have more than one SPI/DSPI port defined is that they assume that you are using the Fubarino SD v1.5 (produced by Microchip) which has three SPI ports in the chip (SPI2, SPI3, and SPI4).

The real action item here is for me to modify the Board_Defs.h and Board_Defs.c files to support both processors, so that if you're using a Fubarino SD v1.5, you get all of the SPI ports on it, but if you're using the v1.4 board you only get one.

In the meantime, you can edit lines 400 to 407 of the chipkit-core\pci32\variants\Fubarino_SD\Board_Defs.h by removing the comment slashes. That will give you DSPI1 on SPI3. You can then add definitions for the last available SPI port (SPI4) if you want - just copy and paste. That will at least get you going for now.

nanoq, 2016-05-17:
The workaround is fine for me, it works.
How looks the future SPI/DSPI implementation for Fubarion SD (v1.5) be?

EmbeddedMan, 2016-05-17:
Hartwig, I'm don't understand your last question. Are you asking how long
before we'll get the 3 SPI/DSPI ports defined in the official chipKIT-core
release? The answer to that is 'hopefully this summer' - we'd like to do
another chipKIT-core release in the next 3 months or so, and I want this to
be fixed by then.

nanoq, 2016-05-17:
Fine Brian.
thx

EmbeddedMan, 2016-05-17:
I'm going to keep this open so it reminds us that we need to make the changes.