chipKIT32/chipKIT-core #225: Wire library missing setClock()

bperrybap, 2016-06-08:
The IDE bundled Wire library includes the function setClock() to allow setting the i2c bus clock frequency. (added both AVR and SAM in IDE 1.5.7)
It would be nice to add this function to the chipkit core to be compatible.
It looks like it wouldn't be hard but would require adding a function to DTWI to do it as currently the setting of the clock rate is done inline in beginMaster() and it looks like you can't just call that again with the frequency from an API function in Wire.

JacobChrist, 2016-07-29:
Something like this:

void twi_setFrequency(uint32_t frequency)
{
  ptwi->ixBrg.reg = (F_CPU/frequency/2) - (((F_CPU / 10000000) * 26) / 25) - 2;
}

bperrybap, 2016-07-29:
I'm not familiar enough with the h/w to know.
But setClock() is a function referenced from the Wire object which is in the  TwoWire class.
i.e. Wire.setClock(frequency);

Are there mutiple i2c busses? Can each of the run at a different clock or they all tied to same clock?

if there are multiple i2c buses each with its own master clock, then clock setting support will need to be pushed into the DTWI class so that each instance can support its own clock.

However it is implemented, the function setClock() needs to be in the TwoWire class so it can be referenced from the Wire object in order to be compatible with the arduino.cc Wire library.

Mejdi2017, 2017-10-04:
In October 2017, Wire.setClock() seem still not implemented?
Is there any plan to implement it in the near future?

majenkotech, 2017-10-05:
I was actually looking at implementing this the other day, however the way that Wire just calls DTWI methods, and DTWI sets the clock when it initially starts master mode, and Wire doesn't, it just seemed too hard to get it to work. So I gave up and switched to using DTWI directly instead.

Mejdi2017, 2017-10-05:
Did it work?

I see that DTWI is another library of Arduino15\packages\chipKIT\hardware\pic32\1.4.3\libraries\
I see that on line 147 of DTWI.h there is the lines:

    typedef enum
    {
        FQ0Hz       = 0,
        FQ100KHz    = 100000,
        FQ400KHz    = 400000,
        FQ1MHz      = 1000000
    } I2C_FREQ;

I guess I need to modify the library here...
Pardon my newbeness but could you walk me through it? ^^

bperrybap, 2017-10-05:
I don' think it is that bad.
I think Jacob's solution is close.
What is needed is a function in DTWI.cpp DTWI class that takes a clock frequency and sets the i2c clock for the DTWI object using the code as noted by Jacob.
Then you should modify DTWI::beginMaster(freq) to use that function to set the clock rather than do it in line.
Then add the setClock() function in WIre.cpp to call the new DTWI function that sets the clock.

Of course in setClock() you have to sanity check the clock frequency passed in and covert it to the appropriate I2C_FREQ enum without going over the requested frequence.

It doesn't seem that bad.

bperrybap, 2017-10-05:
OK,
So I worked out a simple solution that seems to work with my hd44780 lcd library package with PCF8574 chips.
It is very minimal and only requires changes to Wire.h (to add the function to the class) and Wire.cpp for the actual code.
It adds a setClock() function by calling di2c.endMaster() and then calling di2c.beginMaster() with the closest supported frequency.
There are some potential issues here as the Official documentation says some processors supports other rates than the PIC32 core supports.
The main issue being what to do when frequencies lower than 100kHz are requested? like the 10kHz low speed mode.
The goofy Arduino developers defined this function as a void so there is no way to return any kind of error exception status.
To avoid a non working i2c bus, this code will use 100Khz for anything above zero up tot 100Khz. So it will exceed the requested rate when low speed mode is requested.
For all other rates, it will pick a the closest rate that is equal to or lower than the requested rate.
Here is the sample code that seems to work ok for me:
```
// sets i2c clock
void TwoWire::setClock(uint32_t clockFrequency)
{
DTWI::I2C_FREQ freq;

	di2c.endMaster(); // shut down the i2c interface

	// pick closest frequency without going over.

	if(!clockFrequency)
	{
		freq = DTWI::FQ0Hz;
	}
	else if (clockFrequency < 400000L)
	{
		freq = DTWI::FQ100KHz;
	}
	else if (clockFrequency < 1000000L)
	{
		freq = DTWI::FQ400KHz;
	}
	else
	{
		freq = DTWI::FQ1MHz;
	}

	di2c.beginMaster(freq); // restart interface at the closest frequency supported.
}
```

Yes the code is hard coded for a single DTWI object but so is all the other code in the Wire library.

One thing that would be nice would be to have a define that could be used to tell if setClock() exists.
I think it would be reasonable to use a define like what is in the SPI header used to indicate newer capabilities.
So it would be great to add this define to Wire.h
`#define WIRE_HAS_SETCLOCK`

This is useful for libraries and examples like my hd44780 library as it self adjusts to many different cores and versions of the IDE and having a define to indicate setClock() capability in the pic32 core/MPIDE would make using the setClock() function in the examples much easier to auto detect.

--- bill

bperrybap, 2017-10-05:
In terms of storing and remembering a previous clock rate, I think it could probably be argued either way.
While I generally agree that this is a good idea and makes the code more robust, it can't solve some of the issues, given that there are many issues and gotchyas around using setClock() including other cores like the AVR core.

There are some issues with flipping between master and slave with the existing pic32 Wire library.
(I'm looking at this from the perspective of using the Wire library and Wire object vs the DTWI code)

In the pic32 core, the only way beginMaster() is called from the Wire library is if the beginCount is zero
(This goes back to the multiple initialization issue/error from multiple calls to begin() - I reported a long time back)
If beginCount is nonzero Wire will not call beginMaster() or beginSlave() from begin()
The only way to get beginCount to go back to zero after begin() has been called either for master or slave mode, is to call end() which resets beginCount back to zero.
However, end() is not guaranteed to exist since it is conditionally compiled in based the ENABLE_END define.

The AVR Wire library doesn't have a beginCount so anybody can call begin() at any time to change between master and slave modes. (which also resets the i2c clock back to its default in master mode)
As of 1.6.6 the AVR library has an end() function but it isn't documented in the API on the Arduino Wire library page: [Arduino Wire library](https://www.arduino.cc/en/Reference/Wire)
And the AVR Wire code does require that end() be called in order to reinitialized the i2c h/w with begin() even if not changing between modes.

So flipping between master and slave using the Wire library starts to get really messy particularly if trying to create portable code for multiple cores.
- On pic32, you must call end() before you use begin() to change modes, but end() is not guaranteed to exist.  
- On AVR end() is also not guaranteed to exist since it didn't exist until 1.6.6  
- On some cores like AVR you can just call begin() without calling end() to do the flip - but you can't do that on pic32.  
- On some cores like AVR, calling begin() for master mode, even without calling end(), will reset the master clock back to its default.  
- pic32 requires that you call end() before using begin() to change modes, which means it can only be done if it is built with the ENABLE_END define turned on.

Given that the Wire library API does not have a way to set the clock when begin() is called for master mode, I think the only safe way to initialize the i2c interface using the Wire library is to call begin() and then setClock() to set the clock to what you want.

However on AVR, it is still messy since any call to begin() for master mode will reset the clock back to its default. i.e. one library calls begin() and setClock() and then another library calls begin() which sets the clock back to its default.

But I'm not sure that having the Wire library "remember" the last clock set using setClock() for the clock rate for the next begin() is the answer.
It seems like having begin() be smart enough not to re-inialize the h/w until end() is called (like what the pic32 core does) is good enough.
That way, begin() works the same way every time. It initializes the h/w for master mode using a default clock the fist time it is called. All other calls to begin() are ignored until end() is called.
Then to change the master clock setClock() can be called.
It keeps everything simple and consistent.
But like I said I think it could be argued either way.

If the Wire object "remembers" the last clock set, then future begin() calls would use the last clock set whenever begin() was called again.
I think this makes the most sense on a core like the AVR which does not have the beginCount so that if multiple libraries call Wire.begin() the master clock rate set by setClock() is not lost.
But since the pic32 core Wire code doesn't have this issue, it may not be necessary to "remember" the previous clock as setClock() can be called after begin() and there is no issue of it being lost by other libraries calling begin()

majenkotech, 2017-10-05:
Yeesh. What a horrible mess.

I think Wire is long due for an overhaul in the way SPI was - with various per-peripheral profiles so the right clock speed can be applied to the right peripheral - or rather the highest clock rate that satisfies *all* attached peripherals is selected to be used (you can't use a 1MHz clock on a bus with a peripheral that can't work at more than 100kHz - it'd get confused).

bperrybap, 2017-10-05:
I agree Wire is a mess. In the AVR Wire library implementation,
between Wire and twi code there are 5 buffers for data.

But lots of stuff in Arduino is a mess. The original Arduino team was very green when it came to s/w architecture, coding, and development/release practices. And it shows. IMO what really hurt them was that for the first few years, they very much had a rejecting NIH (not invented here) mindset so they wouldn't take or accept comments or even code from anyone outside their small inner circle.

Back on topic....
Baring a re-design of the Wire code and some new Wire API functions,  any suggestions for something better than the above code for how to slip in a setClock() API function in the existing code that works at least as well existing setClock() functions in other cores?

majenkotech, 2017-10-05:
I can think of no better way for the time being. Certainly as a stop-gap solution your proposed one seems the best way at the moment - until Wire gets "fixed"...

And yes, adding the #define is a good call.

bperrybap, 2017-10-06:
ok, should I create a pull request?

EmbeddedMan, 2017-10-06:
Please! By all means. Matt and I will review it and then merge it in.

*Brian

bperrybap, 2017-10-09:
Ok, so I'd like to toss in a little bit more discussion before I create the pull request.
Currently, there is no way to tell if the setClock() is successful or what the actual clock rate is since setClock() returns a void.
I'm wondering if we should try to resolve that at this point in time as well.
Here are the two things I'm considering:
- changing setClock() to return a uint32_t which is the actual i2c clock rate.
The caller can use this return value to see if is clock rate to tell what happened.
- adding a getClock() which returns a uint32_t which is the current i2c clock rate.

Adding getClock() is easy as there is not API compatibility issues.
The change of setClock() to return a value is a messy issue since it is altering the API vs extending it.

For getClock() I would suggest the
define WIRE_HAS_GETCLOCK be added to indicate the presence of getClock()

It would also probably be wise to add a define for indicating that setClock() is returning a value vs a void.

Thoughts?

majenkotech, 2017-10-09:
I don't see the return change as an issue. Since it is currently void it will be being ignored and no return ever stored anywhere in any sketches. Changing to another type won't change that as the returned value will still be ignored. 

It only really matters when you are dealing with virtual functions and polymorphism, which is what happened with write() in the Print class. The return type changed and broke inherited classes. Since nothing inherits from Wire that I know of, and setClock isn't virtual anyway, it won't have any effect.

Yes, it could be useful to know if the return value is available so you can choose to examine it or ignore it depending on the API in use, but few people would assign it anyway since it's not a standard thing in the API...

majenkotech, 2017-10-09:
One thing you could do with setClock is return the *previous* clock rate not the newly assigned one. That way you can easily switch to a new rate and back again with minimum function calls.

bperrybap, 2017-10-09:
I had thought about returning the old clock frequency prior to the setClock(), but I thought it seemed a bit odd not to return a "status" that indicated the result of the setClock() request.
It seemed odd to have to use another API call to see the status of an API call.
i.e. if you want to see if the setClock() actually worked or what it did, you would have to call getClock()

The behavior for intermediate or unsupported clock values is not currently defined in the Wire API.
i.e. what is the result for requesting values that are not 100,000 or  400,000 or 1,000,000 ?
My thought (and it was reflected in the code above) was the setClock() would try to get as close as possible without going over for the rates it supports.
If that is the case, then setClock() may pick a clock that is different than what is requested.
That is part of why I preferred having setClock() returning the clock rate of the i2c interface.
It lets you know the result of what setClock() did.
And if you want/need to know the clock rate you can always call getClock() to get it.

It isn't a big deal either way if getClock() is added to the API, as it can be used to determine the clock before/after setClock() including if setClock() remains as a void function.

The return type of setClock() can be an issue, since library code that may want to use setClock() might have to know if it can get a return value from setClock() or whether it is a void since it could be installed on systems with different versions of the Wire library and be used on different cores.
i.e. it is a backward compatibility issue for new code that wants to get clock status information when run on older cores or non pic32 cores that do not support this capability.

From a 3rd party library or sketch implementation stand point, I think it would be easier (less conditional code) to not have to deal with the getClock() return value complexity and simply use getClock() (if it exists) to get the clock rate information whenever it is desired/needed and available.

Maybe to keep things simple and minimize deviation from the IDE Wire library:
- leave setClock() as a void
- add getClock() that can be used to determine the current state of the i2c clock

Maybe I should bring this up over on the Arduino developers email list?
To see if there is any interest or support for these types of changes/updates.

thoughts?

majenkotech, 2017-10-09:
> The return type of setClock() can be an issue, since library code that may want to use setClock() might have to know if it can get a return value from setClock() or whether it is a void since it [...]

Only new library code. It is impossible for any old library code to want to use a return value since there has never been a return value to use. And new library code can use `#ifdef WIRE_SETCLOCK_RETURNS_INT`

bperrybap, 2017-10-09:
Sure. But what I'm saying is that for new library code that wants/needs to run on the old and the new WIRE library code as well as other cores that do not have these capabilities, the code might be simpler if it only checks for WIRE_HAS_GETCLOCK vs having to check and deal multiple defines and capabilities and environments.

WIRE_SETCLOCK_RETURNS_INT doesn't seem like a good indicator since what we've been discussing is returning either an unsigned it of either the previous clock or the current clock.
I'd be more infavor of WIRE_SETCLOCK_RETURNS_CLOCK or WIRE_SETCLOCK_RETURNS_OLDCLOCK depending on what it returns.

I'm also assuming that eventually I can convince the Arduino boys that add a getClock() but at this point I'm not sure I'll be able to convince them to change setClock() from being a void.
I guess I need to bring this up over on the developers list and get some feedback.

majenkotech, 2017-10-09:
Well, yes, I only chose that name as an example. The actual name would be up to you.

I'm wondering, though, what the actual reasoning is for this. There are only a couple of valid settings for a clock, and you can't really "fail" to set the clock. Sure, getting the current clock could be useful, but the clock "failling" to set would be like a variable "failing" to assign...

    a = 3;
    if (a != 3) { failed(); }

I'm not sure how it can be construed as "failing" unless it just checks the validity of the value you specify - in which case your "closest match" is going to catch it anyway. If you want to see if you got the right clock speed you can just call getClock().

bperrybap, 2017-10-09:
The return value from setClock() doesn't really matter if getClock() exists.

I was looking at this from the perspective of trying to create an API that is stable across all cores and implementations. My concern was that a setClock() implementer (on some other core) might be hardnosed and fail to set/change the clock to the requested clock rate if it wasn't an exact value supported vs picking the closest one. If the previous clock is returned, you end up seeing the "previous" clock which is also still the current clock, so there is no indication that the clock rate sent to setClock() was rejected and you end up always having to call getClock() see if it worked and what it selected (if you care).

Also some cores (like the AVR core) support more than just a few limited clock rates but those rates including the maximum rate supported can vary depending on F_CPU.
i.e. a 8Mhz AVR will use a 500k clock when 1M is asked for.
And by returning the clock picked can the caller can then tell what clock rate was actually selected and is being used.

Also, I can't remember which core, but I have seen code (it was a s/w i2c library) that picks the closest rate even if the picked clock is HIGHER than the requested one. To me this is really dumb. I argued with them but they refused to change it. I think in their case one of the common numbers couldn't be done and the resulting clock was a bit higher than the requested value and they didn't want to drop down to lower rate.
But using other arbitrary rates could end up with a rate that was significantly higher than what was requested.

All this was initially driven by this thread:
http://forum.arduino.cc/index.php?topic=504005.0
Granted this thread was started by what appears to be a newbie that has no clue what he is doing,  but it does bring up the point that currently there is no way to tell the clock rate setClock() picked or the clock rate that is being used.

So there is quite a bit of potential variation in setClock() implementations that can cause a need to see the actual clock rate selected.
So I thought that "while I'm in there" adding setClock() to the pic32 core, I could also try to help resolve this issue was well.
Overall,  I think adding getClock() is a good solution that solves the issue, as then the application/library can do whatever it wants since it can get the pre and post setClock() clock rates.
And it is simple enough that implementers will tend to get it right and consistent across all the implementations.

--- bill

majenkotech, 2017-10-09:
Ah, so it's not really relevant directly for chipKIT (since our core behaves-ish), but it would provide full compatibility with other cores if/when they adopt these proposed changes. Gotcha.

And of course no one wants to be the first to adopt change ;)

But someone has to start the ball rolling, and we're nothing if not progressive, so go for it I say...

bperrybap, 2017-10-09:
One final thing.
Should I  put the setClock() and getClock() code into the DTWI code with wrappers in Wire?
That way users of DTWI can get the functionality.

Or just keep as functionality in Wire but not DTWI ?

majenkotech, 2017-10-09:
Wire. Users of DTWI can already set the clock when they begin a master session. Since the compatibility is for Wire there is no need to duplicate what you can already do in DTWI.

ricklon, 2017-10-11:
Where's the best documentation for using DTWI?

EmbeddedMan, 2017-10-16:
Other than the examples and reading the code, I think there isn't much.

bperrybap, 2017-10-16:
Well, it turns out it isn't as easy it would seem. There are complications.
The Wire library is just a wrapper for DTWI. It doesn't touch h/w.
getClock() is easy to implement, but it can not be done in Wire unless Wire is given access to the h/w which is probably not a good idea since it really isn't necessary as DTWI should provide all the needed services for Wire. (IMO, Wire should be just a convenience wrapper to provide Wire API compatibility)

While I have created a getClock() implemented in DTWI, that  Wire::getClock() can call, the issue is that to be useful with using the Wire version of setClock(), which takes a frequency argument in Hz, the Wire version of getClock() needs to return an actual clock frequency in Hz; however, DTWI currently uses enums to set the clock rate not actual frequencies. (yes the enum values are the frequency but technically, they are not the same as frequencies)
For now, the getClock() I created in DTWI, reads the h/w and back calculates the i2c clock rate and returns the clock rate as a uint32_t. However, while that is what Wire getClock() needs, it is inconsistent with DTWI beginMaster().
So from an API perspective, I'm not sure the best way to implement this.

I really don't want to change things to have Wire touching h/w.
But Wire really does need the clock as a frequency in Hz.
I'm not familiar with all the pic32 processors and the various frequencies they could run at, but I'm assuming there are cases where you might ask for FQ100KHz and not get exactly 100k or FQ400KHz and not get exactly 400k,
Also, there are cases where Wire::setClock() picked the closest clock possible clock which ended up beign different than what was asked for.
So for these reasons, it is best to read the h/w and back calculate the actual frequency and return the actual current i2c clock rate in Hz in Wire::getClock(), but I believe that DTWI should to be the entity to get/calculate the frequency.

So DTWI needs to be the one that reads the h/w and returns the current clock rate in hz.
However, using Hz instead of I2C_FREQ enum is inconsistent with the parameter used in DTWI::beginMaster() which is an enum.
Inconsistencies like this in an API really bother me, but I don't know how to work around it since Wire::getClock() needs to return the actual frequency in Hz and some cases it may not match any I2C_FREQ value so DTWI can't return a I2C_FREQ enum value back to Wire.

Maybe the function in DTWI just needs a different name other than getClock() to be clearer?
getClockFreqHz() 
getMasterFreqHz()
getMasterClockHz()

Any comments/suggestions?

majenkotech, 2017-10-17:
How about a `getClock()` function in DTWI which returns both?

    I2C_FREQ getClock(uint32_t *actualFrequency = NULL) {
        uint32_t theRealFrequency = whateverIsNeededToGetTheClockFrequency;
        if (actualFrequency != NULL) {
            *actualFrequency = theRealFrequency;
        }
        codeToFindTheEnum {
        }
        return theEnum;
    }

Then you can:

    I2C_FREQ clock = dtwi.getClock();

and you get the ENUM.  You can then also do:

    uint32_t freq;
    dtwi.getClock(&freq);

and you get the actual frequency.

Or of course combine the two:

    uint32_t freq;
    I2C_FREQ clock = dtwi.getClock(&freq);

By using a pointer and giving a default of NULL you are effectively making the parameter optional. You could do the same thing with overloading too, if you wanted, though it complicates matters.  Most people wouldn't want the actual frequency when they're working direct with DTWI, so the return value of the ENUM makes sense. On the rare occasions where the actual frequency is wanted it can be dumped into the optional uint32_ pointer parameter for you to then use as you see fit.

It's one function call for two jobs. Nice and efficient, simple, and pretty much transparent.

bperrybap, 2017-10-17:
I didn't think of that. I really like it. That this the way I'll go as it seems to solve everything.

bperrybap, 2017-10-17:
one final thing, Currently in DTWI functions, the master only functions have the string "Master" in their name and the slave only function have the string "Slave" in them i.e. beginMaster(), beginSlave() etc...
getClock() is a master only function.
For consistency should the DTWI getClock() function be named getMasterClock() ?

majenkotech, 2017-10-17:
No real need. The concept of clock in slave mode is non existent. You wouldn't get confused as there is nothing to get confused with.

bperrybap, 2017-11-14:
Ok. So I'm ready to do the commit(s) and pull request.
Just some guidance as to what you guys want.
There are mods to both DTWI and Wire libraries.
DTWI added a getClock() method and Wire added setClock() and getClock()
I also have an example called Master_setClock for the Wire library to demonstrate setClock() and getClock()

The questions:
Do you want these updates on a branch?
If so, do you want the example handled separately from the other updates as
as a separate branch, 2nd pull request, additional commit,  or can everything including the example be just one pull request with a single commit for everything?

Note: as much as I would have liked the Wire getClock() to return the actual clock frequency, to be compatible with the Arduino.cc Wire.getClock() it can't since they declared it as a void.

Here is the new Wire example:
```
// Wire Master_setClock example

// Demonstrates use of the Wire library
// setClock() & getClock() methods.
//
// Wire.setClock(clockFrequencyHZ);
//	returns - nothing
//
// clockFrequencyHZ = Wire.getClock();
//	returns - actual Master mode clock frequency
// 	Note: if 0, Master mode is non operational
// 
// Displays all supported i2c clock rates by
// attempting to set every clock rate from 0 to 1.5M in 10k increments.
// For each attempted clock rate it prints out the actual clock rate 
// being used.
//

// Created 13 Nov 2017

// This example code is in the public domain

#include <Wire.h>

void setup()
{
	Serial.begin(9600);
	Wire.begin();

// Wire.h will set these defines when setClock() & getClock() exist
#if defined(WIRE_HAS_SETCLOCK) && defined(WIRE_HAS_GETCLOCK)
	for(uint32_t f = 0; f < 1500000; f += 10000)
	{
		Wire.setClock(f); // set i2c clock bit rate, if asked
		Serial.print("Set clock: ");
		Serial.print(f);
		Serial.print(" Actual clock: ");
		Serial.println(Wire.getClock());

	}
#else
	Serial.println("Core does not support Wire.setClock()/Wire.getClock()");
#endif
}

void loop()
{
}
```

majenkotech, 2017-11-14:
I usually work in a fresh branch in my own fork, then create a PR from that branch to the master branch on the master fork.

We don't really have branches on the master - certainly not for small things like this. We made one when attempting to incorporate Harmony for the USB stack (which was a complete abortion and triggered me writing one from scratch that is far superior), but that's really all.

You can keep it all as one PR.

EmbeddedMan, 2017-11-14:
I concur with Matt on this. Clean branch of your own fork for your changes,
then a PR from there to the master branch of chipKIT-core works really well.

Thanks again for doing this - I ran into a problem with an LCD graphics
library recently which required I2C to compile (even though I wasn't using
it for the touchscreen) and because of our missing setClock() it wouldn't
compile without modification.

It's fine to put the examples in the same branch and PR as your code
changes.

*Brian

bperrybap, 2017-11-15:
One more thing.
The Arduino IDE documentation for Wire.setClock() [https://www.arduino.cc/en/Reference/WireSetClock](url)
 mentions a 10k low speed mode.
Also, I've noticed that there are some Arduino projects for some soft radio devices that need an initial message sent at 10k.
[https://forum.arduino.cc/index.php?topic=368479.0](url)
[https://www.allaboutcircuits.com/projects/build-an-arduino-controlled-am-fm-sw-radio/](url)
and even after it is initialized at 10k, it still can't run at 100k but can run at 50k.
So without 10k support, the chipkit core could not talk to this device.
The pic can clearly support 10k, and I've modified the code to support it and tested it with my hd44780 lcd library.
If you don't want the 10k support, I'll remove it, otherwise, I'll include it as part of this PR.
Also, if you also want 50k or any other rates, I can add that in as well as part of this PR.
(Yeah, I know this is feature creep, but it is kind of a "while I'm in there" kind of thing)

BTW,
After working with the code and playing with setting/getting i2c clock rates, I am not a fan of using an enum for clock frequencies.
IMO, several things would be simpler if everything used an actual frequency in Hz.
While the Wire library hides all this by allowing the use of actual frequencies, 
It seems that using an enum has created an artificial limitation of what the hardware can really do and in some cases like this soft radio example, it prevents a user from being able to use the device unless they go in patch the Wire and DTWI code to support other frequencies.