chipKIT32/chipKIT-core #232: SoftwareSerial does not compile for FubarinoSD

JacobChrist, 2016-07-05:
Shouldn't chipKIT bot catch this?

Build options changed, rebuilding all
In file included from C:\Users\Jacob Christ\Documents\Arduino\SoftwareSerialTest\SoftwareSerialTest.ino:24:0:

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:68:16: error: '_IOPORT_PC' was not declared in this scope

     { _BV(14), _IOPORT_PC}, // CN0  - RC14

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:69:16: error: '_IOPORT_PC' was not declared in this scope

     { _BV(13), _IOPORT_PC}, // CN1  - RC13

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:70:16: error: '_IOPORT_PB' was not declared in this scope

     { _BV( 0), _IOPORT_PB}, // CN2  - RB0

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:71:16: error: '_IOPORT_PB' was not declared in this scope

     { _BV( 1), _IOPORT_PB}, // CN3  - RB1

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:72:16: error: '_IOPORT_PB' was not declared in this scope

     { _BV( 2), _IOPORT_PB}, // CN4  - RB2

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:73:16: error: '_IOPORT_PB' was not declared in this scope

     { _BV( 3), _IOPORT_PB}, // CN5  - RB3

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:74:16: error: '_IOPORT_PB' was not declared in this scope

     { _BV( 4), _IOPORT_PB}, // CN6  - RB4

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:75:16: error: '_IOPORT_PB' was not declared in this scope

     { _BV( 5), _IOPORT_PB}, // CN7  - RB5

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:76:16: error: '_IOPORT_PG' was not declared in this scope

     { _BV( 6), _IOPORT_PG}, // CN8  - RG6

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:77:16: error: '_IOPORT_PG' was not declared in this scope

     { _BV( 7), _IOPORT_PG}, // CN9  - RG7

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:78:16: error: '_IOPORT_PG' was not declared in this scope

     { _BV( 8), _IOPORT_PG}, // CN10 - RG8

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:79:16: error: '_IOPORT_PG' was not declared in this scope

     { _BV( 9), _IOPORT_PG}, // CN11 - RG9

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:80:16: error: '_IOPORT_PB' was not declared in this scope

     { _BV(15), _IOPORT_PB}, // CN12 - RB15

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:81:16: error: '_IOPORT_PD' was not declared in this scope

     { _BV( 4), _IOPORT_PD}, // CN13 - RD4

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:82:16: error: '_IOPORT_PD' was not declared in this scope

     { _BV( 5), _IOPORT_PD}, // CN14 - RD5

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:83:16: error: '_IOPORT_PD' was not declared in this scope

     { _BV( 6), _IOPORT_PD}, // CN15 - RD6

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:84:16: error: '_IOPORT_PD' was not declared in this scope

     { _BV( 7), _IOPORT_PD}, // CN16 - RD7

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:85:16: error: '_IOPORT_PF' was not declared in this scope

     { _BV( 4), _IOPORT_PF}, // CN17 - RF4

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:86:16: error: '_IOPORT_PF' was not declared in this scope

     { _BV( 5), _IOPORT_PF}, // CN18 - RF5

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:87:16: error: '_IOPORT_PD' was not declared in this scope

     { _BV(13), _IOPORT_PD}, // CN19 - RD13

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:88:16: error: '_IOPORT_PD' was not declared in this scope

     { _BV(14), _IOPORT_PD}, // CN20 - RD14

                ^

C:\Users\Jacob Christ\Documents\Arduino\hardware\chipkit-core\pic32\libraries\SoftwareSerial/SoftwareSerial.h:89:16: error: '_IOPORT_PD' was not declared in this scope

     { _BV(15), _IOPORT_PD}, // CN21 - RD15

                ^

exit status 255
Error compiling.

EmbeddedMan, 2016-07-05:
Currently chipKIT bot is not yet building the sketches. That's a planned
improvement, for when Adam gets time to get it working properly. It is
working part-way, but not all the way yet.

*Brian

JacobChrist, 2016-07-07:
I think this is the same issue as #213

EmbeddedMan, 2016-07-25:
It is, and I just merged the PR that fixes it. All SoftwareSerial examples now build on Fubarino SD.

closed 2016-07-25 by EmbeddedMan