chipKIT32/chipKIT-core #235: Missing pins_arduino.h variant defines/constants

bperrybap, 2016-07-13:
The current pic32 variant files Board_defs.h (which are the equivalent to the Arduino.cc pins_arduino.h files) are missing a few of the Arduino defines & constants.

These are missing: LED_BUILTIN, SDA, SCL.

It is very important that LED_BUILTIN be a #define vs a const integer since existing code uses it for conditional compilation.

It look likes they can easily be created from existing defines.
```
#define LED_BUILTIN PIN_LED1
static const uint8_t SDA =  _DTWIO_SDA_PIN
static const uint8_t SCL = _DTWIO_SCL_PIN
```

It is very unfortunate the the Arduino.cc guys used const values for nearly all of their arduino pin values vs defines as it makes it difficult in cases like this to detect if a constant exists at compile time.

eric-wieser, 2016-11-16:
> It is very important that LED_BUILTIN be a #define vs a const integer since existing code uses it for conditional compilation.

Another option is to use [the following trick](http://stackoverflow.com/q/636595/102441)
```C++
#define LED_BUILTIN LED_BUILTIN
const uint8_t LED_BUILTIN = PIN_LED1;
```

bperrybap, 2016-11-16:
@eric-wieser that is a great idea. That gives the best of both worlds. A cpp define as well as a compile time constant.
@JacobChrist , would it make sense to update the core again to use this instead?

And what about the SDA defines?
I'm thinking it would be good to use for those as well.

JacobChrist, 2016-11-16:
@bperrybap I don't fully understand the consequences of having both a define and a const uint8_t.  Also it would seem like the const uint8_t would need to be defined before the #define.

Agreed, what about the SDA defines?

bperrybap, 2016-11-16:
The define doesn't have to be before or after the const definition.
The preprocessor will simply replace the string with its value which just happens to be same as the define name. The compiler will then see the value. So the define is effectively a NOP as far as the code is concerned.
For example in the case were both forms of LED_BUILTIN existed (define and const), any code that uses LED_BUILTIN would have the string "LED_BUILTIN" replaced with "LED_BUILTIN"  by the pre-processor and the compiler would then see LED_BUILTIN in the code and use the LED_BUILTIN const definition.

In terms of the SDA & SCL defines & definitions they simply don't exist in the chipkit core.
All the Arduino cores have these symbols so they should be created in the chipkit core for each board variant.

eric-wieser, 2016-11-17:
> Also it would seem like the const uint8_t would need to be defined before the #define.
>
> The define doesn't have to be before or after the const definition.

It depends. Putting the define before is safer, because it stops any previous definition from messing up the declaration.

rmcolbert, 2017-02-08:
Arduino also defines board names for conditional compiling.
One example:
#define _VARIANT_CHIPKIT_MAX32_

I'm making a cross-variant compatible robotic arm firmware and things like changing the analogWriteResolution is helpful but doesn't exist on many of the AVR boards. For the DUE, it was simple:
```
  #if defined(_VARIANT_ARDUINO_DUE_X_)
  analogWriteResolution(12);  // for Due
  #endif

```
It would be nice to have for the ChipKIT boards.

EmbeddedMan, 2017-02-08:
Robert - that sounds like a great idea. Can you file a separate issue for
adding a set of chipKIT variant defines that you can use in your sketches?
That way we can track is separately from the missing pins issue.

*Brian