chipKIT32/chipKIT-core #237: SoftwareSerial TX hangs after first byte on PPS based PIC32s

EmbeddedMan, 2016-07-25:
Tested with chipKIT-core v1.2.1, on Fubarino Mini, SoftwareSerialExample sketch. After correctly sending the first "H" of "Hello, world?", the processor appears to hang. No more bytes come out, and USB dies.

EmbeddedMan, 2016-07-25:
My assumption is that this is not just a Fubarino Mini thing, but will affect all PPS based PIC32s. The Fubarino SD (and other non-PPS PIC32s) appear not to have this problem.

JacobChrist, 2016-07-25:
We definitely have seen things hanging Fub Mini but they seem related to the Rx side. Is your Rx pin floating or pulled up? If the Rx pin is floating your asking to hang.

EmbeddedMan, 2016-07-26:
Jacob,

I've tried it both ways - floating and driven. Same problem happens both
ways.

I'm not sure why floating should be a problem - you don't actually start
looking at the RX pin to see if there are edges there until you call
available() I think.

*Brian