chipKIT32/chipKIT-core #240: Wire.requestFrom(address, quantity, stop) missing in chipKIT-core 1.2.1

JacobChrist, 2016-07-29:
As of Arduino 1.0.1, requestFrom() accepts a boolean argument changing its behavior for compatibility with certain I2C devices.

If true, requestFrom() sends a stop message after the request, releasing the I2C bus.

If false, requestFrom() sends a restart message after the request. The bus will not be released, which prevents another master device from requesting between messages. This allows one master device to send multiple requests while in control.

The default value is true.

see: https://www.arduino.cc/en/Reference/WireRequestFrom

bjimmy77, 2020-05-06:
Trying to use some libraries that reference Wire.requestFrom. It is expecting to be able to use 3 argumnets, but the chipKit Wire.h only supports 2. It seems the chipKit Wire library is out of date.

JacobChrist, 2020-05-06:
> Trying to use some libraries that reference Wire.requestFrom. It is expecting to be able to use 3 argumnets, but the chipKit Wire.h only supports 2. It seems the chipKit Wire library is out of date.

Do you have time to update it, that would be great?

majenkotech, 2020-05-13:
My initial thoughts are that it would be as simple as making [this line](https://github.com/chipKIT32/chipKIT-core/blob/master/pic32/libraries/Wire/Wire.cpp#L204) optional. DTWI is a bit of a black box to me though.