chipKIT32/chipKIT-core #249: SPI driver broken for pps pic32

ricklon, 2016-08-19:
The example code is not working.
```
#include <SPI.h>

const int slaveSelectPin = 30;
char x;

void setup() {
  Serial.begin(115200);
  delay(5000);
  pinMode (slaveSelectPin, OUTPUT);
  SPI.begin(); 
}

void loop() {
  digitalWrite(slaveSelectPin,LOW);
  Serial.println(SPI.transfer(x), DEC);
  digitalWrite(slaveSelectPin,HIGH); 
  delay(250);
  x++;
}

```

EmbeddedMan, 2016-08-20:
Well, it is actually, but not on the pins we say it is. :-( 

The above sketch works just fine, if you use  pins 3 (SCK1), 19 (SDI1) and 18 (SDI1). This means that the 'default' SPI port (when using the normal SPI library) is using PIC32 SP1, also known as DSPI0.

The second SPI port on the Mini (PIC32 SPI2, which is 4 (SCK2), 27 (SDI2), 29 (SDO2)), is accessible via DSPI1.

So this is all very confusing. I checked back all the way to chipKIT-core v1.0.1, and all of the above is still true. This means that at the very least our FB Mini Wiki needs an update, and probably the pinout diagram as well.

ricklon, 2016-08-20:
I update the wiki yesterday according to the variant file. Can you confirm
the update is correct?

EmbeddedMan, 2016-08-20:
I re-updated the wiki last night so that it matches reality. I just change
which SPI port we call 'deault'.

*Brian

EmbeddedMan, 2017-03-04:
And after all of the above, the Fubarino Mini documentation was still broken until today. I think I fixed it all now though.

However, I just noticed that the SD library still uses the SPI2 pins. For it's bit-banged SPI port. <sigh>