chipKIT32/chipKIT-core #252: uC32 PWM register problems

bbbnh, 2016-08-23:
I'm pretty new to Arduino-type programming and chipKIT (having only done a bit of python previously), and I'm having trouble making sense of some compile errors I'm running into while playing with [this explanation of advanced PWM setup](http://chipkit.net/changing-pwm-frequency-bit-resolution/). After a bit of poking around, I tried just copying the code from the page into a new file and compiling it, and it gave me the same errors that I'd been running into before, so it seems like there might me something wrong aside from my inexperience with this code.

I'm using the most recent versions of the Arduino IDE (1.6.11) and the chipKIT core (using the board manager link as explained in the chipKIT wiki). I'm working with a uC32 rev. B, and have not attempted to upload any version of this to the board yet. I have tested the board with a few basic blink-/fade-type tests (including a bitbanged PWM implementation).

Has this functionality been broken in recent updates, was it deprecated or never implemented in chipKIT core (if that's a valid question), or am I missing something simple that would make everything work correctly?

I've included a link to a gist that I put together earlier with the basic file I tested, the output from the compiler, and a similar basic explanation of the situation.

https://gist.github.com/bbbnh/5a38ca9b17c4f8e7b5bb86f0f6ded1f3

EmbeddedMan, 2016-08-23:
My quick top of the head guess is that the article you read up on chipKIT.net is written during the time period when we were including the plib.h library in our distrobution (back in the MPIDE days). This library file is no longer included due to licensing issues, and I -think- it included the defined constants you are looking for. (or it included files that did)

The question is then this "In a post-plib.h world, where can one find these constants?"

I'm not quite sure. We no longer include any files in our chipKIT-core that have these constants in them. You can of course add them to the top of your sketch, but that requires you to dig into the datasheet and understand the bits and bytes of the registers you're trying to manipulate.

The 'right' way is to re-write the example code to use the register definitions and bit defines that we have available to us now via the compiler header file. For example, the uC32 uses a PIC32MX340F512H processor, thus the p32mx340f512h.h file (located in C:\Users\bschmalz\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc\pic32mx\include\proc\ under Windows) contains all of the registers and their individual bits that you can use in your code. 

So for example, instead of the line
T2CONCLR = T2_ON;         // Turn the timer off
you would write
T2CONbits.ON = 0;         // Turn the timer off

All of the other lines in the example will need similar translation.

My apologies that this isn't more straightforward.

majenkotech, 2016-08-23:
From your gist, I think this is probably equivalent for the modern compiler:

````C++
void setup() {
  T2CONbits.ON = 0;         // Turn the timer off
  T2CONbits.TCKPS = 3;        // Set prescaler
  TMR2 = 0;                 // Clear the counter
  PR2 = 4096;               // Set the period
  T2CONbits.ON = 1;         // Turn the timer on
}

void loop() {
  OC1R = 2048;              // Load initial value into duty cycle register
  OC1RS = 2048;             // When a period finishes the contents of OC1RS is loaded to OC1R
  OC1CONbits.OCM = 0b110;
  OC1CONbits.OCTSEL = 0;
  OC1CONbits.ON = 1;
}
````

All these registers are in the timer and OC manuals:

* http://ww1.microchip.com/downloads/en/DeviceDoc/61105F.pdf
* http://ww1.microchip.com/downloads/en/DeviceDoc/61111E.pdf

bbbnh, 2016-08-23:
Thanks @EmbeddedMan and @majenkotech for your quick responses and for the info; that example compiles correctly and seems like it might be a bit easier to wrap my head around, so I guess that's this issue resolved for the moment. If I can make sense of it, I might have to write a very basic library to make this stuff a bit more user-friendly. I guess I've got a lot of learning to do!

EmbeddedMan, 2016-08-23:
Sharon,

This is a tough one. There are lots of code samples and tutorials out there
that use code from the plib days that won't compile on chipKIT-core. We
don't want to just delete all of that old stuff (since much of it is not
under our control, so we can't delete all of it) because it's valuable and
we need all of the examples we can get.

We also can't go back and fix all of it, since there isn't developer time,
nor a good way of knowing that you've got all of it.

So I see us as having 2 choices:

1) We do what we're doing now, which is to wait until people have problems
with it, and then guide them to using the newer register syntax.
or
2) We put plib.h back in chipKIT-core so that all of these examples will
work forever.

*Brian

bbbnh, 2016-08-23:
I guess I jumped the gun on closing this. For what it's worth, I found that article by searching for "uC32 chipkit pwm" on Google, so if possible, a note on that article indicating that it's outdated (and ideally linking to an updated article) would be a good start toward getting things up to date. It seems like using this for PWM could be made relatively straightforward; would it help if I were to put together a basic wrapper library to get the ball rolling?

majenkotech, 2016-08-23:
I have started work on adding functions to make all this redundant into the core.

analogWriteResolution() and analogWriteFrequency().

bbbnh, 2016-08-23:
That would be excellent! Would you be willing to put a wild-guess ETA on a basic implementation, or should I study up for now and switch over when the update comes along?

majenkotech, 2016-08-23:
There may be a PR later today if we're lucky.

bbbnh, 2016-08-23:
I was expecting to hear "maybe in the next week or two," so that's awesome news. Will analogWriteFrequency() support setting an arbitrary frequency within the limits of the hardware, or will it be a choice of fractions of the clock speed?

Conceptually, how much of [this](http://www.righto.com/2009/07/secrets-of-arduino-pwm.html) applies to the chipKIT hardware as well, e.g. variable timer top limit and/or 'phase-correct PWM' option?

Thanks for your help and all your work!

majenkotech, 2016-08-23:
Arbitrary frequencies :)

It's just started working as I'd like. A little more testing though...

majenkotech, 2016-08-23:
PR #253 - grab a copy and have a play.

bbbnh, 2016-08-26:
I guess I'll add this thought here as well: would we be able to set different pins to different frequencies? My (limited) understanding of the concepts leads me to believe that this ought to be possible, based on Table 1 on page 4 of [this document](http://www.mouser.com/ds/2/268/61143H-42553.pdf) (Timers/Capture/Compare = 5/5/5). It looks like [Teensy 3.x devices support analogWriteFrequency(pin, freq)](https://www.pjrc.com/teensy/td_pulse.html#frequency) which might actually work for a group of pins rather than single pins, but as far as I can tell, each PWM pin on the uC32 (3, 5, 6, 9, 10) has its own timer.

bbbnh, 2016-08-26:
I got my answer in PR #253, so I guess I can close this now for real.

closed 2016-08-26 by bbbnh