chipKIT32/chipKIT-core #254: The EEPROM library does not work with WiFire boards in chipKIT-core 1.2.1

JacobChrist, 2016-08-24:
Tested with both with WiFire EFC and EFG silicon.

JacobChrist, 2016-08-24:
Just thinking out loud here: The constant: _EEPROM_PAGE_COUNT in Deeprom.c is defined in System_Defs.h as such:

``` c++
#if defined(__PIC32MX1XX__) || defined(__PIC32MX2XX__)
	#define	_EEPROM_PAGE_SIZE	256		// In 32bit words
	#define	_EEPROM_PAGE_COUNT	4
#else
	#define	_EEPROM_PAGE_SIZE	1024	// In 32bit words
#define	_EEPROM_PAGE_COUNT	1
#endif
```
The following is from the PIC32MZ EF family manual:
Note: In PIC32MZ EF devices, the Flash page size is 16 KB (4K IW) and the row size is 2 KB (512 IW).

I don't know if this is causing the problem but it looks like a problem.

JacobChrist, 2016-08-24:
Another possible issue, WiFire ADC uses ADC.  Though I'm not using the ADC in my simple example this comment was found in flash.c:

	// M00TODO: When DMA operations are supported in the core, need
	// to add code here to suspend DMA during the NVM operation.

JacobChrist, 2016-08-24:
Another issue.  From Section 52 (Flash w/support for Live Updates) it indicates the unlock key is:
``` c++
 NVMKEY = 0x0;
 NVMKEY = 0xAA996655;
 NVMKEY = 0x556699AA;
```
The current unlock key is:
``` c++
 NVMKEY = 0xAA996655;
 NVMKEY = 0x556699AA;
```

Anyone know if we can just add the NVMKEY = 0x0; or should we wrap it in an ifdef for the MZ processors?

JacobChrist, 2016-08-24:
Couple more notes:  In flash.c _doNvmOp there is a note about a needed 6us delay after writing to the NVMCON before unlocking memory.  I didn't see anything in either section 5 or 52 manuals about this being needed.  Additionally writing to this register is not mentioned in the section 52 manual so it may not be needed for the EC/EF parts and may actually be detrimental.  Another ifdef around this for MX parts?

I don't have time to test any of these findings tonight, dinner calls.

JacobChrist, 2016-08-25:
In 52.6 ERROR CORRECTING CODE (ECC) AND FLASH PROGRAMMING it states that word programming is not allowed if FEC is enabled.  I can't imagine that it would be turned on by default but I haven't verified one way or the other.

majenkotech, 2016-08-25:
NVMKEY = 0x0;

That is used to clear the key before unlocking. It's common to do it. It doesn't affect the actual unlocking, but it does ensure that it is actually locked before trying to unlock it, and that there is nothing odd in the key register before doing the unlocking that could affect the unlocking.

majenkotech, 2016-08-25:
Ok, the problem we have here is actually two-fold.

First the EEPROM library needs to know the page size, which is simple enough:

````C
#if defined(__PIC32MX1XX__) || defined(__PIC32MX2XX__)
    #define _EEPROM_PAGE_SIZE   256     // In 32bit words
    #define _EEPROM_PAGE_COUNT  4
#elif defined(__PIC32MZ__)
    #define _EEPROM_PAGE_SIZE   4096
    #define _EEPROM_PAGE_COUNT  1
#else
    #define _EEPROM_PAGE_SIZE   1024    // In 32bit words
#define _EEPROM_PAGE_COUNT  1
#endif
````

Secondly the EEPROM flash memory is in the wrong place. At the moment it is located in KSEG0 which is not right. That is cached, which means that any writes to the EEPROM, while they actually work, won't be seen by the current sketch until either the board is reset or the cache block drops out the end of the cache and gets re-read.

So the EEPROM memory needs moving to KSEG1 to bypass the cache. That means changes not only to the core chipKIT-application-COMMON-MZ.ld but also to the linker scripts for the individual boards / chips. 

Basically all references to kseg0_eeprom_addr need changing to kseg1_eeprom_addr and the address changing from 0x9... to 0xB...

Yes, the "kseg0" bit is just a label, but naming it kseg0 and storing it in kseg1 seems a bit silly to me and can only cause confusion later on down the line.

majenkotech, 2016-08-25:
My test program:

````C++
#include <EEPROM.h>

void setup() {
    Serial.begin(115200);
    Serial.printf("Current EEPROM 0 content:    0x%02X\r\n", EEPROM.read(0));
    uint8_t r = analogRead(A0);
    Serial.printf("Setting EEPROM 0 content to: 0x%02X\r\n", r);
    EEPROM.write(0, r);
    Serial.printf("Verify EEPROM 0 content:     0x%02X\r\n", EEPROM.read(0));
    Serial.println("Reset board to try again.");
}

void loop() {
}
````

And the output after making those changes:
````

Current EEPROM 0 content:    0x18
Setting EEPROM 0 content to: 0x6B
Verify EEPROM 0 content:     0x6B
Reset board to try again.
Current EEPROM 0 content:    0x6B
Setting EEPROM 0 content to: 0x5E
Verify EEPROM 0 content:     0x5E
Reset board to try again.
Current EEPROM 0 content:    0x5E
Setting EEPROM 0 content to: 0x38
Verify EEPROM 0 content:     0x38
Reset board to try again.
````

majenkotech, 2016-08-25:
pr #256 created for you to test with.

JacobChrist, 2016-08-26:
I tested pr #256 with a WiFire and it is working.  I suppose we should leave this open until its merged with the core.

JacobChrist, 2016-08-30:
This mostly works in 1.3.1, there is a one-off error that is allowing writes to location 0x200.  The address writable should be 0x1FF.

```
0x000	00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 
0x010	10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 
0x020	20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 
0x030	30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 
0x040	40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 
0x050	50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 
0x060	60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 
0x070	70 71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 
0x080	80 81 82 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 
0x090	90 91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F 
0x0A0	A0 A1 A2 A3 A4 A5 A6 A7 A8 A9 AA AB AC AD AE AF 
0x0B0	B0 B1 B2 B3 B4 B5 B6 B7 B8 B9 BA BB BC BD BE BF 
0x0C0	C0 C1 C2 C3 C4 C5 C6 C7 C8 C9 CA CB CC CD CE CF 
0x0D0	D0 D1 D2 D3 D4 D5 D6 D7 D8 D9 DA DB DC DD DE DF 
0x0E0	E0 E1 E2 E3 E4 E5 E6 E7 E8 E9 EA EB EC ED EE EF 
0x0F0	F0 F1 F2 F3 F4 F5 F6 F7 F8 F9 FA FB FC FD FE FF 
0x100	01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 10 
0x110	11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 
0x120	21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 
0x130	31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F 40 
0x140	41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F 50 
0x150	51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F 60 
0x160	61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F 70 
0x170	71 72 73 74 75 76 77 78 79 7A 7B 7C 7D 7E 7F 80 
0x180	81 82 83 84 85 86 87 88 89 8A 8B 8C 8D 8E 8F 90 
0x190	91 92 93 94 95 96 97 98 99 9A 9B 9C 9D 9E 9F A0 
0x1A0	A1 A2 A3 A4 A5 A6 A7 A8 A9 AA AB AC AD AE AF B0 
0x1B0	B1 B2 B3 B4 B5 B6 B7 B8 B9 BA BB BC BD BE BF C0 
0x1C0	C1 C2 C3 C4 C5 C6 C7 C8 C9 CA CB CC CD CE CF D0 
0x1D0	D1 D2 D3 D4 D5 D6 D7 D8 D9 DA DB DC DD DE DF E0 
0x1E0	E1 E2 E3 E4 E5 E6 E7 E8 E9 EA EB EC ED EE EF F0 
0x1F0	F1 F2 F3 F4 F5 F6 F7 F8 F9 FA FB FC FD FE FF 00 
0x200	02 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x210	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x220	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x230	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x240	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x250	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x260	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x270	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x280	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x290	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x2A0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x2B0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x2C0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x2D0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x2E0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x2F0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x300	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x310	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x320	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x330	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x340	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x350	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x360	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x370	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x380	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x390	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x3A0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x3B0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x3C0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x3D0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x3E0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x3F0	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x400	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x410	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
0x420	FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
[... 204 lines cut ...]

void eeprom_read(char *command)
{
  byte value;
  // there are only MAX_ADDRESS_DEFAULT bytes of EEPROM
  for(int address = 0; address < MAX_ADDRESS_DEFAULT; )
  {
    Serial.print("0x");
    if( address <= 0x0f ) Serial.print("0");
    if( address <= 0xff ) Serial.print("0");
    Serial.print(address, HEX);
    Serial.print("\t");
    for(int i = 0; i < 16; i++, address++)
    {
      // read a byte from the current address of the EEPROM
      value = EEPROM.read(address);
      if( value <= 0x0f ) Serial.print("0");
      Serial.print(value, HEX);
      Serial.print(" ");
    }
    Serial.println();
    delay(100);
  }
}
```

JacobChrist, 2016-08-30:
Actually I'm going to close this one and open a separate issue for the one-off error.

closed 2016-08-30 by JacobChrist