chipKIT32/chipKIT-core #258: EEPROM library: Allow emulated EEPROM to use all of reserved memory.

JacobChrist, 2016-08-26:
The emulated EEPROM uses 1/4 of each reserved 32 bit word.  Improve library to pack the bytes into the words so that we can use all the memory reserved in the linker scripts.  This should be done so that a person upgrading to a new version of the core doesn't over write what is currently stored in EEPROM when gaining this functionality.  For example:

The value stored at EEPROM address 0 would be stored in byte 0 of word 0
The value stored at EEPROM address 1 would be stored in byte 0 of word 1

and so on until then end of memory.  The next byte would be stored in byte 1 of word 0.

majenkotech, 2016-08-26:
Another, more interesting, use for those extra bytes would be to use them to implement a simple 4x wear leveling algorithm.

JacobChrist, 2016-08-26:
Or both and allow user to decide through an exposed parameter in
construction. The MZ part also has the ECC for FEC which looks
interesting.  For the project I am working on it probably would be more
beneficial to have more memory than write cycles.

Jacob

EmbeddedMan, 2016-08-26:
Microchip already has a really nicely done little C library to do EEPROM
emulation in Flash parts (what we're doing) which includes a nice wear
leveling algorithm.

See
http://www.microchip.com/SWLibraryWeb/product.aspx?product=Data%20EEPROM%20Emulation%20for%20PIC

Maybe that would be a better replacement for our EEPROM storage space?

*Brian

majenkotech, 2016-08-26:
You think they'd let us? You remember the hoops we had to go through just to get *part* of Harmony...

JacobChrist, 2016-12-05:
This is actually not an issue, but was a misunderstanding on how the library works.  To get all of the avaible reserved (in linker script) memory use the following code:

``` c++
  EEPROM.setMaxAddress(_EEPROM_PAGE_SIZE - 1); //Try to get as much as we can.
```

closed 2016-12-05 by JacobChrist