chipKIT32/chipKIT-core #259: empty sketch based on Cerebot 32MX7 board fails to compile

jantje, 2016-08-26:
When compiling a empty project based on Cerebot 32MX7 with the arduino eclipse plugin named Sloeber I get an error.
The same error happens in the Arduino IDE 1.6.11
The problem is that the definition of PIN_INT2 has been commented out as "no external pin".
More details in the log below of the Arduino IDE

```
D:\arduino\arduino-1.6.11\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\arduino-1.6.11\hardware -hardware C:\Users\jan\AppData\Local\Arduino15\packages -tools D:\arduino\arduino-1.6.11\tools-builder -tools D:\arduino\arduino-1.6.11\hardware\tools\avr -tools C:\Users\jan\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino\arduino-1.6.11\libraries -libraries C:\Users\jan\Documents\Arduino\libraries -fqbn=chipKIT:pic32:cerebot32mx7 -ide-version=10611 -build-path C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pic32prog.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.0.224-pic32gcc -prefs=runtime.tools.drivers-windows.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\drivers-windows\v2 -prefs=runtime.tools.pic32-tools.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc -verbose C:\Users\jan\Documents\Arduino\sketch_jul28a\sketch_jul28a.ino
D:\arduino\arduino-1.6.11\arduino-builder -compile -logger=machine -hardware D:\arduino\arduino-1.6.11\hardware -hardware C:\Users\jan\AppData\Local\Arduino15\packages -tools D:\arduino\arduino-1.6.11\tools-builder -tools D:\arduino\arduino-1.6.11\hardware\tools\avr -tools C:\Users\jan\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino\arduino-1.6.11\libraries -libraries C:\Users\jan\Documents\Arduino\libraries -fqbn=chipKIT:pic32:cerebot32mx7 -ide-version=10611 -build-path C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.pic32prog.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.0.224-pic32gcc -prefs=runtime.tools.drivers-windows.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\drivers-windows\v2 -prefs=runtime.tools.pic32-tools.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc -verbose C:\Users\jan\Documents\Arduino\sketch_jul28a\sketch_jul28a.ino
Using board 'cerebot32mx7' from platform in folder: C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1
Using core 'pic32' from platform in folder: C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1
Detecting libraries used...
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512L -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_CEREBOT_32MX7_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino   -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\Cerebot_32MX7" "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp" -o "nul"
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512L -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_CEREBOT_32MX7_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino   -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\Cerebot_32MX7" "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MX795F512L -DF_CPU=80000000L -DARDUINO=10611 -D_BOARD_CEREBOT_32MX7_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino   -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\Cerebot_32MX7" "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp" -o "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
"D:\arduino\arduino-1.6.11\tools-builder\ctags\5.8-arduino10/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++"  -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -mprocessor=32MX795F512L -DF_CPU=80000000L  -DARDUINO=10611 -D_BOARD_CEREBOT_32MX7_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino ""  -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\Cerebot_32MX7" "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp" -o "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-gcc"  -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -mprocessor=32MX795F512L -DF_CPU=80000000L  -DARDUINO=10611 -D_BOARD_CEREBOT_32MX7_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino   -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\Cerebot_32MX7" "C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\Cerebot_32MX7\Board_Data.c" -o "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\core\Board_Data.c.o"
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g1 -O2 -Wa,--gdwarf-2 -mprocessor=32MX795F512L -DF_CPU=80000000L  -DARDUINO=10611 -D_BOARD_CEREBOT_32MX7_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino ""  -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\Cerebot_32MX7" "C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32\cpp-startup.S" -o "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\core\cpp-startup.S.o"
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g1 -O2 -Wa,--gdwarf-2 -mprocessor=32MX795F512L -DF_CPU=80000000L  -DARDUINO=10611 -D_BOARD_CEREBOT_32MX7_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino ""  -IC:\Users\jan\AppData\Local\Temp\builded93937a356f4 [... cut ...]
C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32\WInterrupts.c:259:135: note: in definition of macro 'VEC_ENT_CODE'

 #define VEC_ENT_CODE(I,B,P) int c = intmode[I]; IFS0bits.B=0; if (intFunc[I] != 0) (*intFunc[I])(); if(c < CHANGE && c == digitalRead(P)) { IFS0bits.B=1; }

                                                                                                                                       ^

exit status 255
Error compiling for board Cerebot 32MX7.
```

majenkotech, 2016-08-26:
This is a known issue and one that is in my list to create a PR for.

majenkotech, 2016-08-26:
Fixed in PR #260

jantje, 2016-08-26:
Thanks for fixing.
This gets my number of failed tests down with one more.
This was the only one failing for chipKit :-)

JacobChrist, 2017-01-18:
Can we close this issue?

closed 2020-07-03 by jantje