chipKIT32/chipKIT-core #261: empty sketch fails to compile with openscope board

jantje, 2016-08-26:
When compiling a empty project based on openscope  with the arduino eclipse plugin named Sloeber I get an error.
The same error happens in the Arduino IDE 1.6.11
The error comes from a pragma error so I guess something is missing somewhere
More details in the log below of the Arduino IDE
```
D:\arduino\arduino-1.6.11\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\arduino-1.6.11\hardware -hardware C:\Users\jan\AppData\Local\Arduino15\packages -tools D:\arduino\arduino-1.6.11\tools-builder -tools D:\arduino\arduino-1.6.11\hardware\tools\avr -tools C:\Users\jan\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino\arduino-1.6.11\libraries -libraries C:\Users\jan\Documents\Arduino\libraries -fqbn=chipKIT:pic32:OpenScope -ide-version=10611 -build-path C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.drivers-windows.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\drivers-windows\v2 -prefs=runtime.tools.pic32-tools.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc -prefs=runtime.tools.pic32prog.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.0.224-pic32gcc -verbose C:\Users\jan\Documents\Arduino\sketch_jul28a\sketch_jul28a.ino
D:\arduino\arduino-1.6.11\arduino-builder -compile -logger=machine -hardware D:\arduino\arduino-1.6.11\hardware -hardware C:\Users\jan\AppData\Local\Arduino15\packages -tools D:\arduino\arduino-1.6.11\tools-builder -tools D:\arduino\arduino-1.6.11\hardware\tools\avr -tools C:\Users\jan\AppData\Local\Arduino15\packages -built-in-libraries D:\arduino\arduino-1.6.11\libraries -libraries C:\Users\jan\Documents\Arduino\libraries -fqbn=chipKIT:pic32:OpenScope -ide-version=10611 -build-path C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.drivers-windows.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\drivers-windows\v2 -prefs=runtime.tools.pic32-tools.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc -prefs=runtime.tools.pic32prog.path=C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32prog\v2.0.224-pic32gcc -verbose C:\Users\jan\Documents\Arduino\sketch_jul28a\sketch_jul28a.ino
Using board 'OpenScope' from platform in folder: C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1
Using core 'pic32' from platform in folder: C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1
Detecting libraries used...
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MZ2048EFG124 -DF_CPU=200000000UL -DARDUINO=10611 -D_BOARD_OPENSCOPE_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino   -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\OpenScope" "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp" -o "nul"
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MZ2048EFG124 -DF_CPU=200000000UL -DARDUINO=10611 -D_BOARD_OPENSCOPE_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino   -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\OpenScope" "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\tools\pic32-tools\4.8.3-pic32gcc/bin/pic32-g++" -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -fno-exceptions -w -x c++ -E -CC -mprocessor=32MZ2048EFG124 -DF_CPU=200000000UL -DARDUINO=10611 -D_BOARD_OPENSCOPE_ -DARDUINO_ARCH_PIC32 -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino   -IC:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp/sketch "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32" "-IC:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\variants\OpenScope" "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp" -o "C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\preproc\ctags_target_for_gcc_minus_e.cpp"
In file included from C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32/wiring.h:43:0,

                 from C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32/WProgram.h:12,

                 from C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32/Arduino.h:4,

                 from C:\Users\jan\AppData\Local\Temp\builded93937a356f442e64258638d5839ab9.tmp\sketch\sketch_jul28a.ino.cpp:1:

C:\Users\jan\AppData\Local\Arduino15\packages\chipKIT\hardware\pic32\1.2.1\cores\pic32/p32_defs.h:681:14: error: #error Pin definitions have not been configured for your CPU

             #error Pin definitions have not been configured for your CPU

              ^

exit status 255
Error compiling for board OpenScope.

```

JacobChrist, 2016-08-30:
Do you have a link to the board you are using?  I think this is a Digilent product but I can't find it on there web site.

EmbeddedMan, 2016-08-30:
It has not been released yet. We saw a preview version of it at least
year's MASTERs if I remember correctly. I'm not sure when Digilent will be
starting to sell it, but until then our only chance of getting the problem
fixed is for Keith to help us out.

*Brian

majenkotech, 2016-08-30:
Basically the problem is it uses a 124 pin chip. We don't have a PPS defs file for the 124 pin chips. It should be simple enough to create one though - half way between pps/pins_mz_100.h and pps/pins_mz_144.h. Probably take the 144 pin one and delete the pins that aren't relevant.

majenkotech, 2016-08-30:
Fixed in PR #264

JacobChrist, 2016-08-30:
jantje do you work for Digilent, do you have a board?

Jacob

jantje, 2016-08-31:
@JacobChrist 
No I'm not working for Diligent. I am unit testing the project creation and build of the Arduino eclipse plugin named sloeber http://sloeber.io by randomly adding boards from
https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls
I bumped into this one after adding the boards in https://github.com/chipKIT32/chipKIT-core/raw/master/package_chipkit_index.json

JacobChrist, 2016-08-31:
> of-3rd-party-boards-support-urls
> package_chipkit_index.json

Wow, slobber looks like a cool project except for my disdain for Eclipse.
Can it be converted to IntelliJ?

Jacob

jantje, 2016-08-31:
Eclipse has improved the last 4 years. I don't feel like converting to intelliJ.
But he... it is open source ;-)

JacobChrist, 2016-08-31:
Yeah, IntelliJ may not be open.  Its very fast compared to Eclipse though.
If you want open there is the QT IDE, it is amazing!

closed 2020-07-03 by jantje