chipKIT32/chipKIT-core #267: Add -std=gnu11 to c compiler flags in platform.txt

1i7, 2016-09-01:
Hello, I was trying to use plain c library in my sketch (for parsing json http://www.cis.rit.edu/~krz/hacks/jsoncvt/index.html ) and got compilation errors. I am using Arduino IDE with ChipKIT platform plugin. After some investigations I found it can be simply reproduced with new sketch without installing lib:

1. In Arduino IDE (with ChipKIT plugin) create new sketch
2. Click "New tab" on the right (ctrl+shift+N)
3. Type test.c for new file name
4. Add code to test.c tab:

```c
int test() {
  int sum = 0;
  for(int i=0; i<10;i++) {
    sum++;
  }
  return sum;
}
```
5. Select ChipKIT Uno32 board as target
6. (optional, but handy) Enable detailed output for compilation in preferences
7. Press compile, you will receive compilation error:

```bash
"/home/benderamp/.arduino15/packages/chipKIT/tools/pic32-tools/4.8.3-pic32gcc/bin/pic32-gcc"  -c -g -O2 -w  -DARDUINO_ARCH_PIC32 -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -mprocessor=32MX320F128H -DF_CPU=80000000L  -DARDUINO=10609 -D_BOARD_UNO_ -DMPIDEVER=16777998 -DMPIDE=150 -DIDE=Arduino -G1024  -I/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch "-I/home/benderamp/.arduino15/packages/chipKIT/hardware/pic32/1.2.1/cores/pic32" "-I/home/benderamp/.arduino15/packages/chipKIT/hardware/pic32/1.2.1/variants/Uno32" "/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch/test.c" -o "/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch/test.c.o"
/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch/test.c: In function 'test':
test.c:3: error: 'for' loop initial declarations are only allowed in C99 mode
   for(int i=0; i<10;i++) {
   ^
/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch/test.c:3:3: note: use option -std=c99 or -std=gnu99 to compile your code
```

Which means that declaring "int i=0" variable inside for loop (which is quite common for many years) was not implemented in C language until c99 standard.

Same sketch would compile fine, if you change target board from ChipKIT to a kind of Arduino (Arduino Uno for example)

```bash
"/home/benderamp/.arduino15/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.3-arduino2/bin/avr-gcc" -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -MMD -flto -fno-fat-lto-objects -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10609 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   -I/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch "-I/home/benderamp/.arduino15/packages/arduino/hardware/avr/1.6.12/cores/arduino" "-I/home/benderamp/.arduino15/packages/arduino/hardware/avr/1.6.12/variants/standard" "/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch/test.c" -o "/tmp/build0f05a4d52fdf5a63940aca6493fe104b.tmp/sketch/test.c.o"
```

As you can see, Arduino compilation command line uses **-std=gnu11** option for gcc

To edit default compiler options, go to **~/.arduino15/packages/chipKIT/hardware/pic32/1.2.1** and open **platform.txt** file in text editor.

edit **compiler.c.flags=** line, add **-std=gnu11** to the end:

```
compiler.c.flags=-c -g -O2 {compiler.warning_flags}  -DARDUINO_ARCH_{build.arch} -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -std=gnu11
```

Recompile sketch, it should be ok this time.

The minor issue after this change would be new warning during compilation of cpp files:

```
cc1plus: warning: command line option '-std=gnu11' is valid for C/ObjC but not for C++ [enabled by default]
```

It appears because C++ compiler flags in platform.txt is defined via C compiler flags (so this C-specific flag is passed to C++ files as well; for C++ correct flag would be -std=gnu++11):
```
compiler.cpp.flags={compiler.c.flags} -fno-exceptions
```
In Arduino's platform.txt C++ compiler flags are not taken from C compiler flags, so it does not have this warning.

To fix, just copy/paste compiler.c.flag options to compiler.cpp.flag except the new -std=gnu11 one (or replace it with -std=gnu++11 if that makes any sense, Arduino has it)
```
compiler.cpp.flags=-c -g -O2 {compiler.warning_flags}  -DARDUINO_ARCH_{build.arch} -mno-smart-io -ffunction-sections -fdata-sections  -mdebugger -Wcast-align -fno-short-double -ftoplevel-reorder -MMD -std=gnu++11
```

Also, as additional note, if I set C compiler flag to -std=c99 or -std=gnu99 (as compiler message suggests) instead of -std=gnu11, test.c file would compile in this case, but I would receive a number of errors from some system source files, so -std=gnu11 is the only option which works for me.

I think it would be nice, if this option was added to ChipKIT platform plugin by default.

majenkotech, 2016-09-01:
This might be better added to the `recipe.c.o.pattern` (-std=gnu11) and `recipe.cpp.o.pattern` (-std=gnu++11) respectively. That way they don't interfere with each other.

EmbeddedMan, 2016-09-01:
Is there *any* downside to doing this? Are we going to break any code that depends on us not having C11 compatibility turned on?

majenkotech, 2016-09-01:
I doubt it very much. I just experimented in UECIDE and everything compiled quite happily. And since this is the default in the Arduino core these days (AFAIR) it's going to be something we need to have I think.

EmbeddedMan, 2016-09-01:
Cool! Let's do it.