chipKIT32/chipKIT-core #268: zip distribution for mac os x

ricklon, 2016-09-04:
Error when uploading. Substitution is not happening.

```
{runtime.tools.pic32prog.path}/pic32prog -d /dev/cu.usbmodem1411 -b 115200 /var/folders/bx/zg7wx2513knbft_4ch_p5m6c0000gp/T/buildbc65dcc3d94e294db52fa9f290eb84c8.tmp/RC_PulseInTest.ino.hex 
```

EmbeddedMan, 2016-09-04:
Drat. I know what's going on and should be able to fix it shortly.

*Brian

EmbeddedMan, 2016-09-06:
OK, both v1.3.1 and v1.3.0 zip file builds should now be fixed, and should
use the correct location of pic32prog.

Note that their filenames have changed :
http://chipkit.net/wiki/index.php?title=ChipKIT_core#Release_Builds

Please try them out if you can and make sure they work for you.

Thanks for the help in testing!

*Brian

EmbeddedMan, 2016-09-06:
I'm pretty sure this is fixed now - I tested the ZIP file distributions for both v1.3.0 and v1.3.1 on Windows and Linux64, including downloading to different boards, and everything appears to work properly. Please test this on MacOS and if it works, close this issue.