chipKIT32/chipKIT-core #271: EEPROM library: Impliment EEPROM.lenght()

JacobChrist, 2016-09-24:
Although the primary page on arduino.cc doesn't mention an EEPROM.lenght() this page suggests that it may exist and that its there to report the length of EEPROM for devices that have larger EEPROMs.

This issue is to suggest implementing this.

majenkotech, 2016-09-24:
Yes, it does exist. This is all it is in the Arduino core:

    uint16_t length()                    { return E2END + 1; }

We have the same thing defined in cpudefs.h:

````
#if defined(__PIC32MX__) || defined(__PIC32MZ__)

    #define    E2END        0x0fff    //*    4 k of simulated EEPROM
````

Though I guess that's not strictly true is it? The MZ has 16K not 4K doesn't it?

JacobChrist, 2016-09-24:
If we use all 4 bytes of the MZ IW then its actually 64K.

This is nice that its there but its hard coded.  It would be nice if it
used the segment information from the linker to calculate the actual length
on the fly.

Jacob

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com

majenkotech, 2016-09-24:
Also the amount of flash that is reserved for EEPROM does not equal the amount of EEPROM that is available. It's not a straight word0 == EEPROM0, word1 == EEPROM1, but there is some kind of strange addressing / wear levelling system in place that uses the whole page for a small amount of EEPROM.

JacobChrist, 2016-09-24:
Ah, I see.

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com

JacobChrist, 2016-10-03:
I just found a description of the wear leveling in the EEPROM example eeprom_write.  I haven't verified if actual functionality matches described functionality.

There is a functions called getMaxAddress and setMaxAddress.  getMaxAddress() returns 0x200.  The 0x200 value come from this line of code in Deeprom.h:

#define MAX_ADDRESS_DEFAULT		512

I can setMaxAddress as high as 0xFFF (_EEPROM_PAGE_SIZE -1) on a WiFire board.

The funny thing is the EEPROM has that one off issue (error #265) in which you can write to address 0x200.  So this issue may not really be an issue?

The value of E2END on an Arduino Uno that I have is 0x3FF .  It also appears to not be used anywhere and may have been included for compatibility.